espressif/vscode-esp-idf-extension #4: Visual Studio Code hangs when closing integrated-terminal running menuconfig (IDFGH-2443)

Devenda, 2020-01-07:
## Environment
- IDF version (run ``git describe --tags`` to find it): 
   v4.1-dev-1572-g30372f5a4
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
   xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     [Windows]

## Problem Description
When closing the vs code integrated-terminal when it's showing the menuconfig vscode hangs.

This seems to be a vscode issue, but I'm not to sure about that as I'm unable to find the root cause.

I think it may be due to idf.py starting a sub binary (the kconfig-frontend?), but not handling the closing of the terminal correctly?  

I also tested this when running an infinite loop in a python script and also calling a different executable (cmd.exe) from a python script, but both times closing the terminal worked as expected.

All ideas are welcome!

See microsoft/vscode#87840 for more details.

projectgus, 2020-01-07:
(Moving this into VSCode Extension because even though the extension isn't responsible for this, it seems like any fix will probably need to come via the extension.)

SoumeshBanerjee, 2020-01-07:
@Devenda Since we don't have control over the terminal inside the `vscode` directly, for this reason in the __ESP IDF Official VSCode Extension__ we have provided support for the `Menuconfig`, [use that insted](https://github.com/espressif/vscode-esp-idf-extension#idf-gui-menuconfig).

Visit the [extension marketplace](http://marketplace.visualstudio.com/items?itemName=espressif.esp-idf-extension) to know more.

closed 2020-01-21 by SoumeshBanerjee