espressif/vscode-esp-idf-extension #5: kconfiglib's menuconfig rendered badly, barely usable in visual studio code integrated-terminal (Windows). (IDFGH-2442)

Devenda, 2020-01-07:
## Environment
- IDF version (run ``git describe --tags`` to find it): 
   v4.1-dev-1572-g30372f5a4
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
   xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     [Windows]

## Problem Description

After updating to the latest version of esp-idf the menuconfig renders badly in the vscode integrated terminal.
Arrow keys did not work in previous versions either. As a workaround the j and k keys do work.

This is most likely an issue with the vscode integrated terminal, but maybe a workaround/fix is known.
See microsoft/vscode#87838 for more details.
Issue espressif/esp-idf#4067 is also equal to my issue, however using bash is not feasible.
Issue espressif/esp-idf#1969 also seems to be relevant, however removing/cleaning the build dir does not help.

projectgus, 2020-01-07:
(Moving this into VSCode Extension because even though the extension isn't responsible for this, it seems like any fix will probably need to come via the extension.)

SoumeshBanerjee, 2020-01-07:
@Devenda Since we don't have control over the terminal inside the `vscode` directly, for this reason in the __ESP IDF Official VSCode Extension__ we have provided support for the `Menuconfig`, [use that insted](https://github.com/espressif/vscode-esp-idf-extension#idf-gui-menuconfig).

Visit the [extension marketplace](http://marketplace.visualstudio.com/items?itemName=espressif.esp-idf-extension) to know more.

closed 2020-01-21 by SoumeshBanerjee