espressif/vscode-esp-idf-extension #6: Arrow keys do not work in menuconfig (VSCode terminal & Windows Terminal Preview) (IDFGH-1860)

kaofishy, 2020-01-07:
## Environment

- IDF version (run ``git describe --tags`` to find it): 
   v4.1-dev-256-g9f145ff16
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:    Windows 10

## Problem Description

When using `idf.py menuconfig` in a Visual Studio Code integrated terminal or a Windows Terminal Preview terminal, the arrow keys do not work as menu navigation. This does not occur under regular Windows terminal.

### Expected Behavior
Arrow keys navigate the menu items

### Actual Behavior
The menu does not respond to arrow keys

### Steps to repropduce

1. Install Visual Studio Code or Windows Terminal Preview
2. Run `idf.py menuconfig` in a project within the Visual Studio Code integrated terminal or a Windows Terminal Preview session
2. Use arrow keys to navigate the menu

projectgus, 2020-01-07:
(Moving this into VSCode Extension because even though the extension isn't responsible for this, it seems like any fix will probably need to come via the extension.)

projectgus, 2020-01-07:
(Note also that the extension provides a UI config editor, so that may be a better solution than debugging VSCode's interactive terminal support.)

SoumeshBanerjee, 2020-01-07:
@kaofishy  Since we don't have control over the terminal inside the `vscode` directly, for this reason in the __ESP IDF Official VSCode Extension__ we have provided support for the `Menuconfig`, [use that insted](https://github.com/espressif/vscode-esp-idf-extension#idf-gui-menuconfig).

Visit the [extension marketplace](http://marketplace.visualstudio.com/items?itemName=espressif.esp-idf-extension) to know more.

closed 2020-01-21 by SoumeshBanerjee

positron96, 2020-12-21:
For what it's worth, my workaround is to exec `start` from inside VS Code terminal, which opens a normal Windows terminal, and call `pio run -t menuconfig` from there.

JonathanDotCel, 2021-10-09:
Just to add to @positron96 's solution. If VSCode is using a powershell terminal then you must do e.g. `start cmd`
(For the oldschool terminal, just `start` will do the job)

mike6000, 2022-05-17:
I opened a windows terminal with Windows + R and typing cmd
Then running pio run -t menuconfig worked fine for me
(Make sure that pio is in your PATH)

Zeni241, 2023-01-15:
I used `start idf.py menuconfig` , it opened menuconfig in a new detached terminal window  where all the keys works!! I am using powershell as terminal.