espressif/vscode-esp-idf-extension #7: Installation OpenOCD version problem

PiotrParysek, 2020-01-09:
**Environment:**
- VSCode Version: 
1.41.1
- ESP-IDF Version:  (run `cd $IDF_PATH; git describe --tags` to find it):
v3.3.1
- Compiler version (run `xtensa-esp32-elf-gcc --version` to find it):
xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Python Version: `python3 -V`
Python 3.6.9
- Operating System: [Linux/Ubuntu]

**Describe the bug**
During the process of installation (extension installed from the marketplace, with option of downloading the tools), I have encountered a problem with the _wrong_ version of OpenOCD. 
The **Expected** version is: `v0.10.0-esp32-20190708`
The **Found** version displayed: `v0.10.0`

**Possible solution**
To solve the problem, I have edited the `tools.json` file. I changed the attribute: _OpenOCD for ESP32_ and proper (according to the system) attribute: _versions_ (for the newest available version of OpenOCD) and - especially the value for _**name**_

brianignacio5, 2020-01-10:
Hey thanks for your feedback!. It seems there was an issue with updated IDF_PATH before updating the tools.json path.

In the extension folder, there is two backup tools.json (for ESP-IDF >=4.0) and fallback-tools.json (ESP-IDF < 4.0) which are used if ESP-IDF itself doesn't have this file.

For v0.1.2 release these changes will be implemented.

closed 2020-01-16 by espressif-bot

koromkorom, 2020-10-13:
Hi, 

sorry to open this again, but I have the exact same issue. 

Environment:

    VSCode Version: 1.50
    ESP-IDF Extension Version: 0.5.1
    ESP-IDF Version: v4.3-dev-1197-g8bc19ba89
    Compiler version: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
    Python Version: 3.6.9
    Operating System: [Linux/Ubuntu]

At the end of the config process it shows:
Expected v0.10.0-esp32-20200709
Found 0.10.0 

And then an X instead of a check. 

[image]

I could edit some tools.json, but as I understand it, this issue should be fixed entirely.