espressif/vscode-esp-idf-extension #10: Windows asking how do you want to open this file, when running task

jankrejci, 2020-01-16:
When I run some task, windows asks "How do you want to open this file" (means idf.py).

my workaround is to run python command and put the script as an argument, the example for the Clean task
```
"windows": {
    "command": "python",
    "args": [
        "${config:idf.espIdfPathWin}\\tools\\idf.py",
        "fullclean"
    ]
},
```

Windows 10, VSCode 1.41.1, ESP Extension 0.1.2, ESP-IDF Version 4, Python 3.7.3

SoumeshBanerjee, 2020-01-19:
Duplicate of #11

closed 2020-01-19 by SoumeshBanerjee

jankrejci, 2020-01-21:
Hi, I think this issue is not duplicate of #11. The #11 is just regarding OpenOCD environment variables. This issue affect all tasks executing some python script. To solve it you need to install python with py launcher and it is needed to check "Assign files with Python" option. Or the second option is to use the workaround above. The issue is not the problem of the extension itself, but I think it should be at least documented or so.

SoumeshBanerjee, 2020-01-21:
This problem is addressed in #11 and now you should be able to launch from `task.json`, also note that `task.json` can be modify from inside the project to fit your specific environment.