espressif/vscode-esp-idf-extension #13: fresh install of extenstions - menuconfig hangs

mitchjs, 2020-01-17:
i just installed the extension... did all the onboarding setup...
no known errors... everything went well during setup

then created example hello world
and clicked on the launch gui configuration icon

and it did a lot of stuff then hangs
Menuconfig: Loading Initial values...

[image]

mitchjs, 2020-01-17:
`⚠️
CMake Warning at C:/Users/mitchjs/esp-idf/tools/cmake/kconfig.cmake:39 (message):
  Failed to detect version of mconf-idf.  Return code was 1.
Call Stack (most recent call first):
  C:/Users/mitchjs/esp-idf/tools/cmake/build.cmake:160 (__kconfig_init)
  C:/Users/mitchjs/esp-idf/tools/cmake/idf.cmake:43 (__build_init)
  C:/Users/mitchjs/esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:5 (include)

-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
-- Building ESP-IDF components for target esp32

-- Checking Python dependencies...

Python requirements from C:\Users\mitchjs\esp-idf\requirements.txt are satisfied.

⚠️
CMake Error at C:/Users/mitchjs/esp-idf/tools/cmake/component.cmake:222 (message):
  CMake Error at
  C:/Users/mitchjs/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:60
  (include):

    include called with invalid argument:
    Code/Espressif-IDF/hello_world/main/CMakeLists.txt

  Call Stack (most recent call first):

    C:/Users/mitchjs/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:76 (__component_get_requirements)

  

  

Call Stack (most recent call first):
  C:/Users/mitchjs/esp-idf/tools/cmake/build.cmake:404 (__component_get_requirements)
  C:/Users/mitchjs/esp-idf/tools/cmake/project.cmake:337 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!

`

mitchjs, 2020-01-18:
i had a thought... lots of this command line stuff cant handle spaces in path
so my project was at: C:\Users\mitchjs\Documents\Visual Studio Code\Espressif-IDF\hello_world
and it dosnt work
i redid it at
C:\ESP32\hello_world
and it works!

so there is an issue with my long path (really not that long) or the spaces in "Visual Studio Code" part

mitchjs, 2020-01-18:
it just gets worse and worse with this extension
so.. i just tried it all with v3.3.1 release

blink, fails to build
and its menu config fails "ESP-IDF: Menuconfig: Loading initial values..."
screen looks like i posted above

brianignacio5, 2020-01-19:
Hi @mitchjs 

For now menuconfig doesn't work with < v4.0 since there are some new changes we implemented in ESP-IDF for the extension to work we haven't been back ported to v3.3.1.

Make sure you are using the hello world from your currently selected ESP-IDF. There are many breaking changes for v4.0 release which might not work with older ESP-IDF examples .

Can you please share your extension log with us so we can debug what is happening with your setup? Please also add the ISSUE_TEMPLATE information which can help us understand your setup. 

Also share the extension log located in :
Windows %USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-<version>\esp_idf_vsc_ext.log
Mac $HOME/.vscode/extensions/espressif.esp-idf-extension-<version>/esp_idf_vsc_ext.log
Linux $HOME/.vscode/extensions/espressif.esp-idf-extension-<version>/esp_idf_vsc_ext.log

mitchjs, 2020-01-19:
ah.. i started to think that...
and switched back to 4.0 and it worked

I got existing code that was built on 3.3.x 
im "afraid" to switch up to 4.0.x... i guess one day i gotta bite the bullet and go though
it and get it to 4.0.x

closed 2020-01-20 by SoumeshBanerjee

hanyuanh, 2022-03-17:
> i had a thought... lots of this command line stuff cant handle spaces in path so my project was at: C:\Users\mitchjs\Documents\Visual Studio Code\Espressif-IDF\hello_world and it dosnt work i redid [...]

I was using v4.4 and still have this initial value error. After checking your reply and I realized that I was having a space in my path. After moving my stuff to a new path with no space it goes smoothly. Thanks!