espressif/vscode-esp-idf-extension #14: Monitor Fails

mitchjs, 2020-01-18:
when clicking on the monitor icon...
using power shell, doesnt matter if its command prompt... pretty much same error

`PS C:\esp32\blink> set IDF_PATH=C:\Users\mitchjs\esp-idf
PS C:\esp32\blink> C:\Users\mitchjs\.espressif\python_env\idf4.0_py3.8_env\Scripts\python.exe C:\Users\mitchjs\esp-idf\tools\idf.py -p COM7 monitor
Traceback (most recent call last):
  File "C:\Users\mitchjs\esp-idf\tools\idf.py", line 1297, in <module>
    ret = subprocess.call(
  File "C:\Users\mitchjs\AppData\Local\Programs\Python\Python38-32\Lib\subprocess.py", line 340, in call
    with Popen(*popenargs, **kwargs) as p:
  File "C:\Users\mitchjs\AppData\Local\Programs\Python\Python38-32\Lib\subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\mitchjs\AppData\Local\Programs\Python\Python38-32\Lib\subprocess.py", line 1307, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] The system cannot find the file specified`

brianignacio5, 2020-01-21:
Hi !

Can you show us a bit more information about your setup (i.e. ESP-IDF version)?

Can you also share the output of Visual Studio Code "ESP-IDF Build" Output Channel and the extension log located in :
Windows %USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-<version>\esp_idf_vsc_ext.log
Mac $HOME/.vscode/extensions/espressif.esp-idf-extension-<version>/esp_idf_vsc_ext.log
Linux $HOME/.vscode/extensions/espressif.esp-idf-extension-<version>/esp_idf_vsc_ext.log

mitchjs, 2020-01-23:
sure... its something missing in the tools... like not having msys on my machine?
i think its looking WINPTY?

it doesnt matter which version of ESP-IDF, im currently using v4.0(Release)
Windows 10 x64 

`c:\esp32\hello_world>C:\Users\mitchjs\.espressif\python_env\idf4.0_py3.7_env\Scripts\python.exe C:\Users\mitchjs\esp-idf\tools\idf.py -p COM7 monitor
Traceback (most recent call last):
  File "C:\Users\mitchjs\esp-idf\tools\idf.py", line 1298, in <module>
    [WINPTY_EXE, sys.executable] + sys.argv, env=os.environ`

I dont have seem to have a log in "C:\Users\mitchjs\.vscode\extensions\espressif.esp-idf-extension-0.1.3"

which also seems new today?
in 
C:\Users\mitchjs\.vscode\extensions\espressif.esp-idf-extension-0.1.2
i do have a log... ill attach
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/4103928/esp_idf_vsc_ext.log)

mitchjs, 2020-02-13:
ok, i added to idf.py in main()

mitchjs, 2020-02-13:
got it

in VS USER SETTINGS

"terminal.integrated.env.windows": {
        "CHERE_INVOKING": "1",
        "MSYSTEM": "MINGW32",
    },

mitchjs, 2020-02-13:
removed that all good now
and that fixes TASKS too

closed 2020-02-13 by mitchjs

brianignacio5, 2020-02-14:
This should not happend by default in vscode clean install but thanks for the info, will ask users for this vscode configuration for terminal-related issues.

mitchjs, 2020-02-14:
I think it was another extension
I’m sure it was...