espressif/vscode-esp-idf-extension #17: Open Example Project only copies one level deep

Devenda, 2020-01-19:
**Describe the bug**
Creating an example only copies one level of folders.

**To Reproduce**
Open the example menu, choose a project containing multiple levels of folders (eg restful server, touch_pad_read,...) click "Create project using example...".
results in nested folders being treated as files:
[image]

**Expected behavior**
Fully recursively copied folders and files.

**Screenshots**
Leftside: original example in esp-idf folder, rightside: the folder after copy.
[image]

**Environment:**
plugin version 0.1.2
Windows 10

Devenda, 2020-01-19:
Additionally files in the project root are not copied either:
[image]

brianignacio5, 2020-01-20:
Thanks for reporting this issue!

Will fix soon!

closed 2020-01-22 by brianignacio5