espressif/vscode-esp-idf-extension #21: #includes missing in visual studio code

su-Koch, 2020-01-21:
Hey Guys,
I installed the vscode extension. Configuration and download of tools worked pretty well (thanks to the guiding menus). 

Testing the cpp_pthread, includes Paths are not found. All Systems are the current ones.

Console says 
IDF_PATH environment variable is different from inferred IDF_PATH.
What does that mean for me (can not get any IDF_PATH with "set" in vscode) and how to solve this?
I'm a developper in VS2017 but vscode is so different. Appreciate some help here.

Can Please someone support this?

This everthing is so anoying because I switched from eclipse to vscode cause of the include issue.

The ESP32 is really cool but these plugins don't seam to be final versions.

Regards, Manuel

brianignacio5, 2020-01-22:
Hi there ! 

Please can you follow the issue template and show us some steps to replicate your issues?

Regarding your post, currently we depend in Microsoft C/C++ extension to resolve include paths. Please take a look at [C_CPP_CONFIGURATION](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/C_CPP_CONFIGURATION.md) as explained in the [Readme How to use section](https://github.com/espressif/vscode-esp-idf-extension#how-to-use)

When you configured the extension, the ESP-IDF directory path is saved as vscode configuration setting called `idf.espIdfPath` which we use to override IDF_PATH in the visual studio code process (not modifying your system environment variable IDF_PATH in any way). You shouldn't see this warning so if you can tell us which operating system, ESP-IDF version and bash or terminal you are using it will be helpful for us to investigate further.

su-Koch, 2020-01-22:
Hei brian, thank you for your fast reply. How can i see the json config options in vscode set by the idf? For example idf.espIdfPath? I'll provide a stepby step explanation later today. Manuel

brianignacio5, 2020-01-22:
Click F1 -> type _settings_ and you will find an option called Preferences: Open Settings (UI).

Here expand Extensions menu, then click on ESP-IDF and you will see the configuration settings we expose in our extension.

More about visual studio code settings [here](https://code.visualstudio.com/docs/getstarted/settings)

su-Koch, 2020-01-22:
Hei brian, I double checked my settings and got it to work. I use esp-idf v3 now and had to set these options in the project (no idea why):
[image]

Thanks for your help, maybe I've better asked this general question in forum. 
I really apreciate your work. ESP32 is really nice.

Regards, Manuel

SoumeshBanerjee, 2020-01-23:
Possible duplicate of #2 
 
`"includePath": ["${config:idf.espIdfPath}/components/**"],`
`*` is required to search recursivelu for all the components for header/ symbols resolutions

closed 2020-01-23 by SoumeshBanerjee