espressif/vscode-esp-idf-extension #25: Error parser

bitmakk, 2020-01-24:
Hello!

My problem is, that with even the **0.1.3** the VSCode can't highlight me in the Editor (or at the Problems) the errors/warnings given by the output (at build). I tried more settings in the tasks.json file, but none of it worked. 
Can you give me tips to use it properly? 
(I added errors to my code, it fails to build, but the VSCode doesn't recognize it as a problem.
The problem matcher pattern is matched by the output (as I tested in an online tester). )

SoumeshBanerjee, 2020-01-25:
HI @bitmakk 

The `problem-matcher` for IDF Build, flash and other such task is still under development and will be shipped soon in the upcoming versions of extension.

brianignacio5, 2020-02-18:
Di you try configuring your c_cpp_properties.json to use compile_commands.json ?

This is specified in [C/C++ Configuration](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/C_CPP_CONFIGURATION.md) as mentioned in [How to use](https://github.com/espressif/vscode-esp-idf-extension#how-to-use).

As we mention in the README file, the C/C++ extension is used to resolve C headers, which is not part of our extension thereby user still need to configure it.

SoumeshBanerjee, 2020-03-03:
@bitmakk  Adding this feature will take a while, meanwhile you could use the tasks available in task.json or write your own

closed 2020-03-03 by SoumeshBanerjee

SoumeshBanerjee, 2020-05-18:
@holubcsongor This feature request is now released and available in v0.2.2