espressif/vscode-esp-idf-extension #27: Include errors after fresh install

bartoszwalicki, 2020-01-28:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [x] Make sure you have searched for existing bugs and features request before you post an issue.
- [x] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [x] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
After fresh installation of extension (and all required dependencies) and onboarding process (full success) I occurred problem with include error, but only on this line:
`#include "freertos/FreeRTOS.h"`
In any example project there is the same warning. Project builds and flash without any problems.

Here is my `c_pp_properties.json` (nothing changed here)

``` json
{
    "configurations": [
        {
            "name": "Linux",
            "cStandard": "c11",
            "cppStandard": "c++17",
            "includePath": ["${config:idf.espIdfPath}/components/**"],
            "browse": {
                "path": [
                    "${config:idf.espIdfPath}/components"
                ],
                "limitSymbolsToIncludedHeaders": false   
            }
        }
    ],
    "version": 4
}
```

**To Reproduce**

1. Install vscode extenstion
2. Perform onboarding
3. Generate new project based on blink example
4. Error should pops out in Problems pane and in editor

**Expected behavior**
Problems pane should be empty.

**Screenshots**
[image]

**Environment (please complete the following information):**
- OS Name & Version: [Arch Linux 5.4.15-arch1-1 GNOME 3.34.2 ]
- VSCode Version: [e.g. 1.41.1]
- ESP-IDF Version: [e.g. v4.1.0]
- Python Version: [e.g. 3.8.0]

**Additional context**
I' ve downloaded ESP-IDF manually as stated [https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#linux-and-macos](here) and provided path to esp-idf manually in onboarding process.

Here is my `settings.json` if it is relevant:

``` json
{
    "terminal.integrated.fontFamily": "Source Code Pro for Powerline",
    "typescript.updateImportsOnFileMove.enabled": "never",
    "editor.formatOnSave": true,
    "editor.fontWeight": "600",
    "idf.pythonSystemBinPath": "/usr/bin/python3",
    "idf.espIdfPath": "/home/bartosz/esp/esp-idf",
    "idf.toolsPath": "/home/bartosz/.espressif",
    "idf.pythonBinPath": "/home/bartosz/.espressif/python_env/idf4.1_py3.8_env/bin/python",
    "idf.customExtraPaths": ".:/usr/bin:/home/bartosz/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin:/home/bartosz/.espressif/tools/xtensa-esp32s2-elf/esp-2019r2-8.2.0/xtensa-esp32s2-elf/bin:/home/bartosz/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/bartosz/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/home/bartosz/.espressif/tools/openocd-esp32/v0.10.0-esp32-20191114/openocd-esp32/bin",
    "idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"/home/bartosz/.espressif/tools/openocd-esp32/v0.10.0-esp32-20191114/openocd-esp32/share/openocd/scripts\"}",
    "idf.showOnboardingOnInit": false,
    "idf.projectName": "hello-world",
    "idf.port": "/dev/ttyUSB0"
}
```

mehmetcanbalci-Notrino, 2020-01-30:
I got similar errors for the blink.c , after fresh installation. we are in 2020 and still there is no any proper installation for any kind of program.

bitmakk, 2020-01-30:
The problem is with the compiler-dependent header files.
To fix this, you have to set the c/cpp extension to the correct settings.
Press F1, then select C/C++ Edit configurations (UI).
Compiler path: esp-gcc location... (Looks like this: C:\Espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe )
IntelliSense mode: gcc-x64
Include path: 
${workspaceFolder}/**
${config:idf.espIdfPath}/components/**

With these settings I could fix the missing system-includes

bartoszwalicki, 2020-02-02:
@bitmakk It works! Maybe it is good idea to include this line in default project generation settings?

Now I occurred another problem with imports, but I think that it is connected with some C/C++ intellisense cache:
[image]

brianignacio5, 2020-02-14:
This is specified in [C/C++ Configuration](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/C_CPP_CONFIGURATION.md) as mentioned in [How to use](https://github.com/espressif/vscode-esp-idf-extension#how-to-use)

KevinDaddy, 2020-02-15:
I encountered the same problem, and solved by set the right develop kit for cmake extension.
You may not see the right xtensa compiler at first time, don't worry, just select the "Scan for kit" action, you will see the xtensa compiler in the kits list.
All of the above, is based on the configurations of esp-idf vscode extension are set correctly.
Enjoy!

brianignacio5, 2020-02-17:
Similar to #26 we updated the template with [Update PR C_Cpp_properties.json template](https://github.com/espressif/vscode-esp-idf-extension/pull/40/files).

Closing this since issue was solved by reporting user.

closed 2020-02-17 by brianignacio5