espressif/vscode-esp-idf-extension #28: Building ouput format unexpected

syu-source, 2020-02-09:
**Describe the bug**
Esp-idf build Output format contains unexpected spaces and lines.
Here is part of the logs.

```
[1/909] Generating project_elf_src.
c

[2/909] Generating linker s
cript...

[3/909] Generating ../../partition_table/partition-table.bin

[4/909] Building ASM obje
ct esp-idf/f
r
e
e
r
t
o
s/CMakeFiles/__idf_
freer
tos.dir/xtensa
/xtensa_int
r
_asm.S.obj

[5/909] Building ASM
 object esp-idf/xtensa/CMakeFiles/__idf_xtensa.dir/expression_with_stack_xtensa_asm.S.obj

[6/909] Building ASM object esp-idf/xtensa/CMak
eFiles/__idf_xtensa.dir/debug_helpers_asm.S.obj

[7/909] Building C object esp-idf/xtensa/CMakeFiles/__idf_xtensa.dir/eri.c.obj

[8/909] Building C obje
ct esp-idf/xtensa/CMakeFiles/__idf_xtensa.dir/trax.c.obj

[9/909] Building C obj
ect esp-idf/xtensa/CMakeFiles/__idf_xtensa.dir/debug_helpers.c.obj
```

**To Reproduce**
1. Install Esp-idf extension, get onboard, and create a helloworld project.
2. Build (and the building successes.)
3. examine the output log

**Expected behavior**
Building log shows line by line.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
- OS Name & Version: [Win10]
- VSCode Version: [1.42]
- ESP-IDF Version: [4.1]
- Python Version:  [2.7.16]

**Additional context**
Btw, I also have the issue about include Error, like following header file not found in the editor.
#include "freertos/FreeRTOS.h"  

Tried solution mentioned in VSC-229, didn't work out.  
So I hardcoded absolute path in .vscode\c_cpp_properties.json, and finally got it work. 
It looks like that the idf.espIdfPath is not resolved correctly, wired.
The esp-idf extension is correctly configured because the project builds.
```
            "includePath": [
                "${workspaceFolder}/**",
                "D:/workspace/vscode/esp-idf.v4/components/**",
                "${config:idf.espIdfPath}/components/**"

            ],
```

bitmakk, 2020-02-10:
If I remember correctly, the idf.py, and the vscode needs python3 instead of python2.
Maybe you should try it with it, if it helps.

syu-source, 2020-02-10:
Sorry, the python version is 3.7. 
The onboard check of esp-idf extension says all installation test passed and the project builds all right.

bitmakk, 2020-02-10:
I think, the problem can be in src/buid/build.ts file (line 100):
outputToOutputChannel(data: string) function copies the string to a new line in the output, BUT:
I didn't found nowhere clues, that the output from build is line separated. My suspicion is, that it is character-buffered, and gives the newline command in \n character. The evidence for this theory, that in the output there is an empty new line after every "normal" line, which is not here during commandline build. 
The developer should try  this.outputChannel.append(data); instead of appendLine

SoumeshBanerjee, 2020-02-11:
@syu-source You can try out the `.vsix` file generated in the PR #30 
Hopefully this will solve the issue you're facing!

syu-source, 2020-02-11:
It works for the building output.

But the esp-idf extension ICON on the left panel is missing..  though the tooltip showed up.
[image]

SoumeshBanerjee, 2020-02-11:
This is very weird, could you please try restarting the VSCode (because for me its showing the icon).
In case this still don't work, this will hopefully get fixed once we put next version extension in the marketplace

syu-source, 2020-02-11:
After restarting vscode, the icon was still missing. However,  the icon showed itself after I closed the workspace and afterwards.

It's working now.

syu-source, 2020-02-11:
Since it works and I cannot reproduce it. I close this issue here.
Thanks @SoumeshBanerjee  for the quick fix.

PR #30 (2020-02-12): Enhance Build Output (VSC-235)
Fix the build output to new line for some OS/ Terminal Environment
Closes #28
closed 2020-02-11 by syu-source