espressif/vscode-esp-idf-extension #31: Need to update app-template

syu-source, 2020-02-11:
**Describe the bug**
CMakefile.txt is using old components registration API, which doesn't have document support.

**To Reproduce**

Press F1 and type ESP-IDF: Create ESP-IDF project to generate a template ESP-IDF project.

**Expected behavior**
I suppose CMakefile.txt should be using `idf_component_register `instead of `register_component`

**Screenshots**
[image]

**Environment (please complete the following information):**
OS Name & Version: [Win10]
VSCode Version: [1.42]
ESP-IDF Version: [4.1]
Python Version: [3.7]

**Additional context**
https://github.com/espressif/esp-idf-template

bitmakk, 2020-02-11:
The two code is the same, it is just a cmake macro.
There is no "old components registration API", and it is perfect in these way too.

syu-source, 2020-02-12:
It works both ways, yes.
But I could only find document at 
https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html#

I'm moving my project from legacy GNU Make system, and get confused by Cmakefile.txt of the default app template because it doesn't match the official document.

Anyway, I built my project based on examples which is using `idf_component_register `.

PR #36 (2020-02-13): Use new IDFv4 Cmake `idf_component_register` API (VSC-240)
Closes #31
closed 2020-02-13 by SoumeshBanerjee