espressif/vscode-esp-idf-extension #32: Linker Error: failed to parse ESP-IDF/components/esp_event/linker.lf

armandr, 2020-02-11:
while I'm able to easily build and flash the hello_world sample from idf command prompt, the VS Code Extension gives the following error:

`failed to parse C:/Users/r/esp/ESP-IDF/components/esp_event/linker.lf`

detail:
```

[2/5] Generating ld/esp32.project.
ld
FAILED: esp-idf/esp32/ld/esp32.project.ld 
cmd.exe /C "cd /D C:\Users\r\esp\hello_world\build\esp-idf\esp32 && python c:/Users/r/esp/ESP-IDF/tools/ldgen/ldgen.py --config C:/Users/r/esp/hello_world/sdkconfig --fragments C:/Users/r/esp/ESP-IDF/components/esp_ringbuf/linker.lf	C:/Users/r/esp/ESP-IDF/components/spi_flash/linker.lf	C:/Users/r/esp/ESP-IDF/components/esp_wifi/linker.lf	C:/Users/r/esp/ESP-IDF/components/lwip/linker.lf	C:/Users/r/esp/ESP-IDF/components/esp_event/linker.lf	C:/Users/r/esp/ESP-IDF/components/espcoredump/linker.lf	C:/Users/r/esp/ESP-IDF/components/esp32/linker.lf	C:/Users/r/esp/ESP-IDF/components/esp32/ld/esp32_fragments.lf	C:/Users/r/esp/ESP-IDF/components/xtensa/linker.lf	C:/Users/r/esp/ESP-IDF/components/soc/linker.lf	C:/Users/r/esp/ESP-IDF/components/heap/linker.lf	C:/Users/r/esp/ESP-IDF/components/freertos/linker.lf	C:/Users/r/esp/ESP-IDF/components/newlib/newlib.lf	C:/Users/r/esp/ESP-IDF/components/app_trace/linker.lf	C:/Users/r/esp/ESP-IDF/components/esp_gdbstub/linker.lf --input C:/Users/r/esp/ESP-IDF/components/esp32/ld/esp32.project.ld.in --output C:/Users/r/esp/hello_world/build/esp-idf/esp32/ld/esp32.project.ld --kconfig c:/Users/r/esp/ESP-IDF/Kconfig --env-file C:/Users/r/esp/hello_world/build/config.env --libraries-file C:/Users/r/esp/hello_world/build/ldgen_libraries --objdump C:/Users/r/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-objdump.exe"
linker script generation failed for C:/Users/r/esp/ESP-IDF/components/esp32/ld/esp32.project.ld.in
ERROR: failed to parse C:/Users/r/esp/ESP-IDF/components/esp_event/linker.lf
Expected end of text, found 'i'  (at char 0), (line:1, col:1)
ninja: build stopped: subcommand failed.

```

This is a fresh install of VSCode, and the extension.

- OS Name & Version: Win 10
- VSCode Version: 2019
- ESP-IDF Version: 4.0
- Python Version: 3.7 (whichever is installed by the idf)

**Additional context**
prior to this, I was getting errors about missing python packages Pyserial  and pyparsing.  These are present when I open the IDF command prompt but not in a regular cmd.exe, so I pip installed them.

bitmakk, 2020-02-11:
I don't think, that this error is related to the vscode plugin.
The cmake/ninja fails, not the handling of output.

brianignacio5, 2020-02-14:
Please follow our issue template, which specify the necessary information we need to understand your issue.

Have you set up the extension following steps detailed in https://github.com/espressif/vscode-esp-idf-extension#how-to-use ?

KevinDaddy, 2020-02-15:
Try to delete the build directory and build again, I dont why, but it works!

brianignacio5, 2020-02-18:
Since the error and information shown by the user doesn't seem to be an issue with ESP-IDF extension for Visual Studio Code, we are closing this.

closed 2020-02-18 by brianignacio5

nguyentranho, 2020-09-27:
Uninstall pyparsing and reinstall its version 2.2.0 will solve the problem
pip uninstall pyparsing
pip install pyparsing==2.2.0
There are a version conflict between the internal python at $HOME\.espressif\python_env\idf4.2_py3.8_env\Scripts\python.exe and system's python