espressif/vscode-esp-idf-extension #39: IDF Onboarding virtualenv unrecognized arg: --no-site-packages

IKingJohnI, 2020-02-14:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [X] Make sure you have searched for existing bugs and features request before you post an issue.
- [X] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [X] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
Onboarding is trying to execute `virtualenv` with `--no-site-packages`, which has been deprecated for some time now. Latest Windows IDF installer installs a version that doesn't have it any more, causing onboarding to fail. This was just recently fixed in the Windows installer: https://github.com/espressif/esp-idf/issues/4744

**To Reproduce**
Install ESP-IDF using the standard installer (latest 2.3 version). Install ESP-IDF VSCode extension, start onboarding. Onboarding will fail on `Define ESP-IDF tools install directory.` stage.

**Expected behavior**
Onboarding to proceed, virtualenvs to be created.

**Screenshots**
[image]

**Environment (please complete the following information):**
- OS Name & Version: Windows 10, 1904
- VSCode Version: 1.41.1
- ESP-IDF Version: 4.1 I think
- Python Version: 3.7.3

**Additional context**
This was just recently fixed in the Windows installer: https://github.com/espressif/esp-idf/issues/4744

IKingJohnI, 2020-02-14:
Some notes for those who need it:

Line 81 of `src/pythonManager.ts` contains the offending code. 
Change: `${pythonBinPath} -m virtualenv --no-site-packages ${pyEnvPath}`,
To: `${pythonBinPath} -m virtualenv ${pyEnvPath}`,

Follow build instructions on main readme.md. You'll want to build the `.vsix`. It's worth noting that it says to use `yarn run build_vsix` to build the `.vsix`, though this seems to be incorrect. I ran `yarn run package` and got the `.vsix`. Then, uninstall the extension in VS Code if you currently have it installed. Then, click the three dots in the upper right of the Extensions panel. Click `Install from VSIX...`. You know what to do from there. 
[image]

brianignacio5, 2020-02-14:
Thanks for reporting this !

Will be fixed on v0.1.4

closed 2020-02-14 by brianignacio5

mitchjs, 2020-02-14:
is this anything to be concerned with when i do the package

yarn run v1.22.0-esp-idf-extension>yarn run package
**warning esp-idf-extension@0.1.3: The engine "vscode" appears to be invalid.**
$ vsce package --yarn -o esp-idf-extension.vsix
Executing prepublish script 'yarn run vscode:prepublish'...

SoumeshBanerjee, 2020-02-14:
@mitchjs 
> warning esp-idf-extension@0.1.3: The engine "vscode" appears to be invalid.

NO! this is just a warning, just ignore this.

PR #40 (2020-02-14): remove site packages (VSC-245)
Closes #39 

Closes #26