espressif/vscode-esp-idf-extension #44: Wrong version of toolchain installed for IDF v3.3.1

ftab, 2020-02-18:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [x] Make sure you have searched for existing bugs and features request before you post an issue.
- [x] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [x] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
I tried the vscode extension from the marketplace and ran into a bug that's been fixed already, so I built the extension from source and installed that. When selecting IDF v3.3.1 it still downloads the gcc 8 toolchain which doesn't work with 3.3.1.

I have worked around by manually running esp-idf/tools/idf_tools.py after finishing the extension onboarding process, and changing the custom extra paths.

I didn't see any other ways to configure which version of the toolchain to download, but I may have missed them.

**To Reproduce**
Build extension from source and install
Select v3.3.1 during onboarding

**Expected behavior**
It downloads the correct version of gcc and other tools compatible with v3.3.1 by running esp-idf/tools/idf_tools.py install

**Environment (please complete the following information):**
- OS Name & Version: Ubuntu 18.04
- VSCode Version: 1.42.1
- ESP-IDF Version: 3.3.1
- Python Version: 2.7.17

brianignacio5, 2020-02-25:
Hi @ftab I tried building the vsix from the current master in Github and for v3.3.1 the version shown in Downloads is  in fact 5.2.0 of the Xtensa Toolchain.

Probably delete `dist`, `out` and `nnode_modules` and re-run `yarn` to avoid any conflict data.

Please let us know if the problem persist.

ftab, 2020-03-01:
Here's what I see:

[image]

I selected 3.3.1 in the onboarding screen and these are the downloads that are showing. 8.2.0 etc.

So, this is still happening as of 0.1.4

ftab, 2020-03-01:
I think the problem might be here:

https://github.com/espressif/vscode-esp-idf-extension/blob/33ed3b9e631f0adede8071ee3c0fc891c52d7008/src/views/onboarding/Download.vue#L71

I downloaded the esp-idf

https://github.com/espressif/vscode-esp-idf-extension/blob/de488167bc3b7d1cabf6d50135d0b4873d42e5b1/src/onboarding/espIdfDownload.ts#L91

which "set" the idf path

https://github.com/espressif/vscode-esp-idf-extension/blob/c6dbd06593cb05fab63f794a152820c854e74a8e/src/views/onboarding/main.ts#L59-L62

but it did not "save" the idf path

https://github.com/espressif/vscode-esp-idf-extension/blob/33ed3b9e631f0adede8071ee3c0fc891c52d7008/src/views/onboarding/store/actions.ts#L31-L33

https://github.com/espressif/vscode-esp-idf-extension/blob/de488167bc3b7d1cabf6d50135d0b4873d42e5b1/src/onboarding/OnboardingPanel.ts#L93-L97

so the updateIdfToolsManager was never called

https://github.com/espressif/vscode-esp-idf-extension/blob/de488167bc3b7d1cabf6d50135d0b4873d42e5b1/src/onboarding/OnboardingPanel.ts#L236-L240

so the onboarding default values are still being used (which was not aware of the esp-idf version yet and was defaulting to tools.json)

If I quit vscode rather than downloading the wrong toolchain, then I reopen, and open the onboarding again, and select the now existing esp-idf, it does fetch the right toolchain versions, as it calls "saveIdfPath" when I click the "Go to ESP-IDF tools setup"

https://github.com/espressif/vscode-esp-idf-extension/blob/33ed3b9e631f0adede8071ee3c0fc891c52d7008/src/views/onboarding/Download.vue#L40

So I think the fix might be to make "setIdfPath" (after downloading) also call "updateIdfToolsManager", or call "saveNewIdfPath" instead after downloading.

brianignacio5, 2020-03-03:
Can you post more info, for example what is printed in ESP-IDF Output tab?

Also please share your extension log in ~/.vscode/extensions/<espressif.esp-idf-vscode-extension-<version>/esp_idf_vsc_ext.log

boarchuz, 2020-03-08:
I've also experienced this and agree with ftab's diagnosis: idfConf.writeParameter("idf.espIdfPath" updateIdfToolsManager isn't called after downloading esp-idf and setting the new directory (https://github.com/espressif/vscode-esp-idf-extension/blob/81bcc954a25ea179bebe7e8c839f8b50d3bd078d/src/onboarding/espIdfDownload.ts#L93), so if the newly downloaded version has crossed either way pre/post v4.0 from the initial version detected by Onboarding, then the tools json path will be incorrect. 

It can be reproduced by alternately downloading v3.3/v4.0, continuing to "Download ESP-IDF Tools" and noting the version for xtensa-esp32-elf, then going back to Onboarding start and repeating for the other version. Which tools version is incorrect depends on which IDF version the extension detected on Onboarding start.

It only happens when downloading; 'Find ESP-IDF in your system' will correctly call updateIdfToolsManager: https://github.com/espressif/vscode-esp-idf-extension/blob/81bcc954a25ea179bebe7e8c839f8b50d3bd078d/src/onboarding/OnboardingPanel.ts#L96

brianignacio5, 2020-03-09:
Thanks for the feedback, I've got the issue reproduced as well and will fix soon.

PR #54 (2020-03-13): update idf tools, refactor some promise await (VSC-258)
Fix #44  by updating idfTools manager after esp-idf download.

Review await/async statements. 

await for configuration.update() on writeParameter
closed 2020-03-13 by brianignacio5