espressif/vscode-esp-idf-extension #46: Option to select where settings should be saved

KaeLL, 2020-02-20:
Currently, Onboarding works by appending the settings to the user settings, and building the project won't work if those settings are not specifically there.

There should be a way to choose if the settings should be user-wide and workspace-wide. An option for Onboarding and another one for the Command Palette. Consequently, the extension would default to look for workspace settings first before moving on to look at user settings, following the regular VSCode flow of workspace settings overriding user's.

The main reason why this is a must have it's because it makes it easy to work with multiple projects, versions of the toolchain, and versions of ESP-IDF. Another reason is pollution of user configurations, especially if VSCode isn't used solely for ESP32 development. I'm sure there are others, but these are the ones that I can think of right now.

brianignacio5, 2020-03-03:
Thanks for the feedback.

Ideally when you set up vscode configuration settings, there is a preference such as workspace folder settings -> workspace settings -> user settings. Although in the current master we had no such feature yet (currently on review) you should be able to override user settings with workspace ones and those would be used instead. **Isn't that the case ?**

KaeLL, 2020-03-03:
> Isn't that the case ?

No, it's the opposite. User settings should override Workspace settings. Along with the 2 reasons I already mentioned, this is also VSCode's regular flow (User > Workspace).

brianignacio5, 2020-03-03:
> > Isn't that the case ?
> 
> No, it's the opposite. User settings should override Workspace settings. Along with the 2 reasons I already mentioned, this is also VSCode's regular flow (User > Workspace).

https://code.visualstudio.com/docs/getstarted/settings

KaeLL, 2020-03-03:
LoL I actually got them backwards in my last comment. I meant to say Workspace > User. Anyway, thanks.

PR #50 (2020-04-02): configuration save destination (VSC-255)
Fix #46 

Enable the user to select where to save configuration settings.

On vscode commands -> extra step to define where to save configuration settings.

On Onboarding -> dropdown on first window to select where to save configuration settings for the whole onboarding experience.
closed 2020-04-02 by brianignacio5