espressif/vscode-esp-idf-extension #49: Debug feature estimation time

jmpmscorp, 2020-02-28:
Hi.

My esp32 develop is growing in complexity quickly. Till now, I can "debug" programs in old school mode... uart logs. However, I need better and accurate debug tool. Is there an estimation when debug feature will be available in vscode? should I change to eclipse IDE + idf plugin?

Thanks in advance.

brianignacio5, 2020-02-28:
Hi there,

We are currently working on our debug adapter for vscode which will be available soon.

In the meantime you can try to configure your launch.json like:

```
{
    "version": "0.2.0",
    "configurations": [
        {
            "type": "cppdbg",
            "request": "launch",
            "name": "ESP-IDF: Debug",
            "miDebuggerPath": "xtensa-esp32-elf-gdb",
            "program": ${workspaceFolder}/build/${command:espIdf.getProjectName}.elf",
            "environment": [{"name": "PATH", "value": "${config:idf.customExtraPaths}"}],
            "miDebuggerServerAddress": "localhost:3333",            
            "args": [],
            "stopAtEntry": false,
            "cwd": "${workspaceFolder}",
            "environment": [],
            "externalConsole": true,
            "MIMode": "gdb"
        },
    ]
}

```
Certainly our eclipse IDE has better GDB support, so can try it as well.

jmpmscorp, 2020-02-28:
Thank you very much @brianignacio5 

I'll try both suggestions.

closed 2020-02-28 by jmpmscorp

jmpmscorp, 2020-03-16:
Hi again.

Excuse to comment again. Following IDF documentation, I could configure project to debug with Eclipse IDE. However, when I have tried with vscode, nothing happens.

@brianignacio5. I put your suggestion in launch.json. When I click on debug, nothing starts. In esp-idf CMD window, openocd is started succesfully but no vscode connection is shown with debug click. If I open vscode deleveloper tools, console shows this error:

```
ERR Error: Can't read file "c:\Users\JM\.vscode\extensions\espressif.esp-idf-extension-0.1.4\src\kconfig\kconfig.code-snippets" (EntryNotFound (FileSystemError): Error: ENOENT: no such file or directory, open 'c:\Users\JM\.vscode\extensions\espressif.esp-idf-extension-0.1.4\src\kconfig\kconfig.code-snippets')
    at _.doReadAsFileStream (fileService.ts:457) file:///c%3A/Users/JM/.vscode/extensions/espressif.esp-idf-extension-0.1.4/src/kconfig/kconfig.code-snippets
```

I have opend C:/Users/... esp-idf-extension directory and there isn't any "src" folder. Only dist, i18n, media, syntaxes and templates folder exit.

Can you give me any tip?

Thanks in advance.

brianignacio5, 2020-03-19:
This error is not related to the debug part. Will fix but is related to the Kconfig files.

Did you configure your extension using the `ESP-IDF: Configure ESP-IDF extension` ? Please share your `esp_idf_vsc_ext.log`

jmpmscorp, 2020-03-19:
Hi again.

> Did you configure your extension using the ESP-IDF: Configure ESP-IDF extension ?

Yes, I used idf extension home screen to configure it. I specially put attention on configuration and all ticks was marked correctly. Also, I have tried debug in other PC with same result.

> Please share your %USERPROFILE%/.vscode\extensions\espressif.esp-idf-extension-0.1.4\esp_idf_vsc_ext.log content.

There isn't this file. I attach an screenshot where you can see vscode configuration and debugger console. %userprofile%/... directory and cmd window with openocd.

[image]

I clicked many times on debug before screenshot.

As you can see, some screenshots messages could be in spanish. If you don't understand something, tell me.

Regards