espressif/vscode-esp-idf-extension #53: Invalid template-app CMakeLists with IDF version < 4

boarchuz, 2020-03-07:
The CMakeLists.txt for the template-app has been updated for IDF v4.x (https://github.com/espressif/vscode-esp-idf-extension/pull/38) which breaks compatibility with IDF v3.x:

> CMake Error at /.../main/CMakeLists.txt:1 (idf_component_register):
>   Unknown CMake command "idf_component_register".

Perhaps there should be a duplicate template-app with the original v3-compatible CMakeLists.txt? eg. "template-app (for IDF pre-v4.0)".

SoumeshBanerjee, 2020-03-09:
Since the extension aims to be `IDF v4.x +` compatible, thats why the default template project is upgraded to be using the latest version of `IDF v4` API.

That being said, you could still run IDF projects with `IDF version < 4`

In future we will add a template IDF < v4 project

brianignacio5, 2020-03-10:
You could also create from example using `ESP-IDF: Show ESP-IDF Examples` which uses the current `idf.espIdfPath` to get the examples and copy them into your new directory.

closed 2020-03-20 by SoumeshBanerjee