espressif/vscode-esp-idf-extension #58: Support ESP8266 RTOS SDK?

mriksman, 2020-03-11:
Looking for an alternative to Arduino; I need a platform that can compile ESP32 and ESP8266 code without any/many changes to the original code. 
Arduino is NONOS
PlatformIO is the (very) old ESP8266 RTOS SDK 1.5beta5

With ESP8266 RTOS SDK being similar to ESP-IDF, can this extension support ESP8266? That way, I have one environment to program the two different chips.

brianignacio5, 2020-03-11:
At some point, the work of ESP8266 RTOS SDK will be migrated to ESP-IDF as stated here https://github.com/espressif/ESP8266_RTOS_SDK#roadmap 

Unfortunately the extension can't work for ESP8266 RTOS SDK since it is focused mostly ESP-IDF, ESP-IDF tools and CMake. When ESP8266 RTOS SDK is available on ESP-IDF, we will provide support for it in the extension.

closed 2020-03-11 by SoumeshBanerjee

mriksman, 2020-03-11:
@brianignacio5 Thanks. I saw that roadmap - great news.

Doesn't ESP8266 RTOS SDK work with CMake now as well? And idf.py etc?