espressif/vscode-esp-idf-extension #61: Can not start debugger

jankrejci, 2020-03-19:
I have a problem to start debugger. OpenOCD is running, but the debugger stops almost immediatelly after the green arrow is pressed. Unfortunatelly no output is printed in either console. There is probably problem with the ${command:espIdf.getProjectName} variable. I have tried to find workaround and the following is working. Also there is needed to run xtensa-esp32-elf-gdb.**exe** under the windows.

launch.json
```
"miDebuggerPath": "xtensa-esp32-elf-gdb",
"program": "${workspaceFolder}/build/app-template.elf",
"windows": {
                "miDebuggerPath": "xtensa-esp32-elf-gdb.exe",
                "program": "${workspaceFolder}\\build\\app-template.elf",
 },
```

Extension 0.1.4
vscode 1.43.1
python 3.8.1
Windows 10

meltdown03, 2020-03-21:
I have the same problem in Linux with the latest build from the master branch including #57. If I change the `${command:espIdf.getProjectName}.elf` to the actual file name, it works. I can't find a log of the crash.

brianignacio5, 2020-03-24:
Thanks for the feedback. Fixed in #68

meltdown03, 2020-03-26:
The new version 0.2.0 has a few issues with debugging. Clicking "Launch" does not start OpenOCD, even with "mode" set to "auto", or not set at all. That line in launch.json also gives an error that "auto" must be an enum.
https://github.com/espressif/vscode-esp-idf-extension/blob/8c4e10d9e1c4182265dc9c050c0d1bf617745427/package.json#L425 file has "mode" set to "enum", I think it should be "string".

 I can get it to work if I click the task "OpenOCD: Start OpenOCD", then click "Launch", but only if "mode" is not set at all. With the "mode" option set to either option, esp_debug_adapter does not start unless I also start "apapter" manually. 
Maybe OpenOCD isn't supposed to start automatically, but the "mode" option is definitely not working correctly for me.

meltdown03, 2020-03-27:
> I get a toast message the says "OpenOCD Exit with non-zero error code 1".

I had the OPENOCD_SCRIPTS env variable set incorrectly. Now works when "mode" is not set., It opens OpenOCD and Debug Adapter. If I set it to "auto" - connect ECONNREFUSED 127.0.0.1:43474 and neither OpenOCD nor Debug Adapter opens. "manual" does the same. Both work the same If I manually open OpenOCD and "adapter"

meltdown03, 2020-04-02:
I'm having issues withe the new DAP. I set a breakpoint but the code stops at the same line in `pm_esp32.c' everytime. I have to click the correct Thread under Call Stack to get it to show the correct line.
[image]
[image]

PR #68 (2020-04-23): find xtensa toolchain full path fix getProjectName (VSC-272)
Fix #61 by adding a command to resolve xtensa toolchain gdb full path and fix getProjectName command. Updated the previous launch.json template.

Tested on MacOS
closed 2020-04-23 by brianignacio5