espressif/vscode-esp-idf-extension #64: getting charmap

chenbhil, 2020-03-21:
Windows 10.
Python 3.8.2
ESP-IDF Version: 4.0

Opened a new User in my computer(English language instaed of Hebrew on the other two), fresh intalled VSCODE,Git,Python,ESP-IDF.
this works on another computer in the same way I did to this user. what am I missing????
I'm trying to solve this 3 days now, losing hope, losing my mind.

I get this problem when trying to build blink example:
`-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.25.1.windows.1") 
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/ESP32/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/ESP32/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/ESP32/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Users/ESP32/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Users/ESP32/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
Python requirements from C:\Users\ESP32\esp-idf\requirements.txt are satisfied.
⚠️ Traceback (most recent call last):
  File "C:/Users/ESP32/esp-idf/tools/kconfig_new/confgen.py", line 571, in <module>
    main()
  File "C:/Users/ESP32/esp-idf/tools/kconfig_new/confgen.py", line 230, in main
    config = kconfiglib.Kconfig(args.kconfig)
  File "C:\Users\ESP32\esp-idf\tools\kconfig_new\kconfiglib.py", line 653, in __init__
    self._parse_block(None,           # end_token
⚠️   File "C:\Users\ESP32\esp-idf\tools\kconfig_new\kconfiglib.py", line 1632, in _parse_block
    self._parse_block(_T_ENDMENU,
  File "C:\Users\ESP32\esp-idf\tools\kconfig_new\kconfiglib.py", line 1584, in _parse_block
    prev_node = self._parse_block(None,            # end_token
  File "C:\Users\ESP32\esp-idf\tools\kconfig_new\kconfiglib.py", line 1584, in _parse_block
    prev_node = self._parse_block(None,            # end_token
  File "C:\Users\ESP32\esp-idf\tools\kconfig_new\kconfiglib.py", line 1551, in _parse_block
    while self._has_tokens or self._next_line():
  File "C:\Users\ESP32\esp-idf\tools\kconfig_new\kconfiglib.py", line 1186, in _next_line
    self._line = self._file.readline()
  File "C:\Users\ESP32\AppData\Local\Programs\Python\Python38-32\lib\encodings\cp1255.py", line 23, in decode
    return codecs.charmap_decode(input,self.errors,decoding_table)[0]
UnicodeDecodeError: 'charmap' codec can't decode byte 0x8c in position 3651: character maps to <undefined>
⚠️ CMake Error at C:/Users/ESP32/esp-idf/tools/cmake/kconfig.cmake:208 (message):
  Failed to run confgen.py
  (C:\Users\ESP32\.espressif\python_env\idf4.0_py3.8_env\Scripts\python.exe;C:/Users/ESP32/esp-idf/tools/kconfig_new/confgen.py;--kconfig;C:/Users/ESP32/esp-idf/Kconfig;--sdkconfig-rename;C:/Users/ESP32/esp-idf/sdkconfig.rename;--config;C:/Users/ESP32/Documents/blink/sdkconfig;--defaults;C:/Users/ESP32/Documents/blink/sdkconfig.defaults;--env-file;C:/Users/ESP32/Documents/blink/build/config.env).
  Error 1
Call Stack (most recent call first):
  C:/Users/ESP32/esp-idf/tools/cmake/build.cmake:441 (__kconfig_generate_config)
⚠️   C:/Users/ESP32/esp-idf/tools/cmake/project.cmake:337 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!`

SoumeshBanerjee, 2020-03-22:
@chenbhil Whats is happening if you're building the `blink` project from `cmd` directly using `idf.py build` command, are you also getting the same error?
I am suspecting there could be some error with some value received in `kconfig` or somewhere along those lines...

chenbhil, 2020-03-22:
forgot to mention, same output from cmd and from IDE.
Tried doing the same with `release branch of esp-idf v4.0`, nothing.

chenbhil, 2020-03-22:
The computer that MANAGES to build the project has cmd's character page set to 850.
ALL the computers/users that can't build are set to 852.
I tried to build with cmd after doing `chcp 850`, didn't work.

brianignacio5, 2020-03-23:
Another ESP-IDF (using the Eclipse plugin instead) has some similar error here https://esp32.com/viewtopic.php?f=40&t=14502&p=56275

Since the error is from the ESP-IDF side, try using `PYTHONIOENCODING=utf-8` as environment variable before running `idf.py build` and see what happens.

chenbhil, 2020-03-23:
Done that too.
I added it as I add IDF_PATH.
I've heard that this variable needs to be set along with another but I can't find which.

SoumeshBanerjee, 2020-03-24:
This looks like python or idf related issue, raising query in esp32.com forum would be more helpful in that case

projectgus, 2020-03-24:
Hi @chenbhil ,

We've been tracking a similar issue internally where parts of the IDF build & config system breaks if the system locale is not UTF-8 (it seems to be the `charmap` codec in this case).

Can you please try setting a new environment variable, `PYTHONUTF8` with value `1`, and see if the problem goes away? You should be able to configure this in the VSCode project config or as a Windows environment variable (although if you run other Python-based software, setting this variable for all of Windows may also change their behavior.)

Angus

chenbhil, 2020-03-27:
Hello and thank you for the response,

When adding to `{"PYTHONUTF8":"1"....(existing env vars)..}`  `idf.customExtraPaths` in `@ext:espressif.esp-idf-extension`
 I get: `-- Checking Python dependencies...
⚠️ F⚠️ atal Python error: config_init_utf8_mode: invalid PYTHONUTF8 environment variable value`

P.S: does `terminal.integrated.detectLocale` have anything to do with our problem?

Thanks,
Chen

projectgus, 2020-03-30:
Hi @chenbhil ,

That's unexpected, it looks like CPython will throw this error if it sees a value other than "1" or "0" in the variable. But it certainly looks like you set the variable to "1":

https://github.com/python/cpython/blob/3c30a76f3d3c0dcc1fb4de097fa4a3a4c92c0b0b/Python/preconfig.c#L620

brianignacio5, 2020-03-30:
You should probably first try using it in Powershell or cmd.exe outside Visual Studio Code first to see if there is Visual Studio Code issue there.

By the way, additional environment variables should go in `idf.customExtraVars` as explained in the readme.

projectgus, 2020-04-20:
@chenbhil Did you have any success getting Python to recognise the `PYTHONUTF8` environment variable value?

closed 2020-04-26 by github-actions[bot]