espressif/vscode-esp-idf-extension #72: Build task problems in the Problems tab have unresolved filepaths

boarchuz, 2020-03-26:
**Describe the bug**
After running the build task, the problems tab is populated with entries parsed from the output. 
However, clicking one of these results in a VS Code error: 

> Unable to open 'main.cpp': Unable to read file '/../main/main.cpp' (Error: Unable to resolve non-existing file '/../main/main.cpp').

Where the actual filepath is '/home/boarchuz/Projects/MyProject/main/main.cpp'

**To Reproduce**
In a project with any error, "Tasks - Run task" -> "Build - Build project". Click the resulting entry in the Problems tab.

**Solution**
The default problemMatcher config (https://github.com/espressif/vscode-esp-idf-extension/blob/d1f5fd16b3e574a5e95726b93e7660aae915e760/templates/.vscode/tasks.json#L18) has "fileLocation": "absolute" while the output provides relative paths.

Example build output error:
> ../main/main.cpp: In function 'void app_main()':
> ../main/main.cpp:82:7: error: 'some_undeclared_variable' was not declared in this scope
>      ++some_undeclared_variable;
>        ^~~~~~~~~~~~~~~~~~~~~~~~

This can be solved by changing "fileLocation" to the following:
`"fileLocation": ["relative", "${workspaceFolder}/build"],`

The only thing I'm not 100% sure about is if that path is ideal. ie. Which path is the build output's "../main/main.cpp" actually relative to? Using "fileLocation": ["relative", "${workspaceFolder}"] would also work with an altered regexp to handle the "../".

There are many other tasks that seem to use the same problemMatcher config that may be affected too.

**Environment**
Linux Mint 19.3, VS Code 1.43.1, IDF Extension 0.1.4, IDF 'master' (v4.2-dev-701-g0ae960f2f)

SoumeshBanerjee, 2020-03-31:
> The only thing I'm not 100% sure about is if that path is ideal. ie. Which path is the build output's "../main/main.cpp" actually relative to? Using "fileLocation": ["relative", "${workspaceFolder}"] would also work with an altered regexp to handle the "../".

You're correct and we are still investigating if this regexp will handle all the cases or not

boarchuz, 2020-04-01:
It seems that errors in files outside of the project folder (eg. external components) have their absolute filepaths given, while project filepaths are relative.

It's possible to provide an array of problemMatchers to handle different cases. 
Here's my working config now for example:

```
              "problemMatcher": [
                {
                    "owner": "cpp",
                    "fileLocation": ["relative", "${workspaceFolder}"],
                    "pattern": {
                        "regexp": "^\\.\\.(.*):(\\d+):(\\d+):\\s+(warning|error):\\s+(.*)$",
                        "file": 1,
                        "line": 2,
                        "column": 3,
                        "severity": 4,
                        "message": 5
                    }
                },
                {
                    "owner": "cpp",
                    "fileLocation": "absolute",
                    "pattern": {
                        "regexp": "^[^\\.](.*):(\\d+):(\\d+):\\s+(warning|error):\\s+(.*)$",
                        "file": 1,
                        "line": 2,
                        "column": 3,
                        "severity": 4,
                        "message": 5
                    }
                }
            ],
```

PR #86 (2020-05-15): update tasks with relative paths (VSC-308)
Fix #72 

PTAL @boarchuz @SoumeshBanerjee 

Tested a bit with Windows and MacOS, seems to be ok for build and flash tasks.
closed 2020-05-14 by github-actions[bot]