espressif/vscode-esp-idf-extension #74: OpenOCD-CLIENT, and additional functions and requests.

ArTyGill, 2020-03-29:
I want to use my jtag adapter for debugging via the Internet, without a direct connection via usb using port 3333, so I want to be able to debug via OpenOCD-client.
I also want to say that I would like to see the output of information in one window, because after build and flash the data output window does not go between the commands.

SoumeshBanerjee, 2020-03-31:
> I would like to see the output of information in one window, because after build and flash the data output window does not go between the commands.

Merging all the outputs into single output window would create lot of noise, whole idea behind output window is to navigate to them when there are some failures and look at what went wrong in details, we usually extract the minimal relevant information from the whole battery of logs and filter them to UI based popups and notification for the users

> I want to use my jtag adapter for debugging via the Internet, without a direct connection via usb using port 3333, so I want to be able to debug via OpenOCD-client.

@an-dr @brianignacio5 any thoughts ?

ArTyGill, 2020-03-31:
@SoumeshBanerjee 
Yes, there is a lot of information that does not need to be compiled, and it has drawbacks (it is worth reading the output from idf.py), but it would be nice if I had the opportunity to switch between the outputs of the tasks separately and the general stream where the output from all incepts is conducted and which is cleared when each new task is launched.

And I get a python binary path error every time I start monitor, but everything works just fine.

Thanks!

an-dr, 2020-03-31:
> I want to use my jtag adapter for debugging via the Internet, without a direct connection via usb using port 3333, so I want to be able to debug via OpenOCD-client.

Hi! What is OpenOCD client? GDB? Personally I don' t see any problem to expose options to specify address and port for gdb to connect to OOCD in some future release

SoumeshBanerjee, 2020-03-31:
> but it would be nice if I had the opportunity to switch between the outputs of the tasks separately
This was the initial UX but its quite distracting if users keeps building and launching openocd and other debug task which will create a race condition of opening their own output panel and this is very distracting for most of the users

> And I get a python binary path error every time I start monitor, but everything works just fine
If this is a new error or issue you're getting or reporting its worth opening a new issue with full details

ArTyGill, 2020-03-31:
@an-dr I have a raspberry pi with built-in jtag on which I connected esp32 and launched openocd-server and can access it on port 3333, and if you add the openocd client functionality, I can debug through raspberry or from my computer or laptop or other  device over a local area network.

@SoumeshBanerjee yes, that's right it is openocd that creates a lot of noise in the output, and I would like to select only a few items (assembly and firmware) that will need to be output to the general stream.
And
Yes, I will create a separate problem.

SoumeshBanerjee, 2020-03-31:
> I would like to select only a few items (build and flash) that will need to be output to the general stream

Right now vscode don't provide a nice multiplexing api to enable such functionality, that being said we have created an internal ticket for tracking whenever such functionality is available will integrate the same.

ArTyGill, 2020-03-31:
@SoumeshBanerjee Thanks!

brianignacio5, 2020-03-31:
Yes agreed on limited API for Output channels for the moment.

There have been attempts to use Tmux within vscode, for example https://medium.com/@joaomoreno/persistent-terminal-sessions-in-vs-code-8fc469ed6b41 which we could probably try for initial remote debug scenarios.

closed 2020-10-26 by github-actions[bot]