espressif/vscode-esp-idf-extension #75: problems with missing python dependencies

lafar6502, 2020-03-29:
Windows 10
esp-idf 4.2 (master) - installed from git
esp-tools - downloaded and installed separately, v 2.3

Installed ESP-IDF vscode extension but at the end of process i had some error message about missing python dependencies (which was strange because all python dependencies were already installed by the esp tools setup). Fixed this by running pip install command suggested by the error message, and then was able to finish the onboarding successfully, but still get error when trying to build esp-idf projects:

```
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
The following Python requirements are not satisfied:
click>=5.0
pyserial>=3.0
future>=0.15.2
cryptography>=2.1.4
pyparsing>=2.0.3,<2.4.0
pyelftools>=0.22
esp-windows-curses; sys_platform == 'win32'
Please follow the instructions found in the "Set up the tools" section of ESP-IDF Getting Started Guide
⚠️ CMake Error at C:/tools/esp-idf/tools/cmake/build.cmake:271 (message):
  Some Python dependencies must be installed.  Check above message for
  details.
Call Stack (most recent call first):
  C:/tools/esp-idf/tools/cmake/build.cmake:397 (__build_check_python)
  C:/tools/esp-idf/tools/cmake/project.cmake:395 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "C:/espdev/masterpiec2/blink/build/CMakeFiles/CMakeOutput.log".
```
cmakeoutput.log file shows just normal compilation, no error messages

But all the required libraries are installed - they were installed during esp tools setup. And to be sure i reinstalled them. No success.

brianignacio5, 2020-03-30:
**The issue is that the extension is expecting the python executable in ** `idf.pythonBinPath` which is used as python executable for the extension commands. In the `ESP-IDF: Configure ESP-IDF extension` command, we create a python virtual environment and install the required python packages there.

As specified in Step 3 in the [Onboarding](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/ONBOARDING.md) extension configuration, a python virtual environment is created (same as in ESP-IDF Tools installation directory) which is passed to the extension build commands. Please take a look at the readme to properly configure your extension as well the former document which explain what `ESP-IDF: Configure ESP-IDF extension` is doing.

lafar6502, 2020-03-30:
Ok, i think i fixed it - as you say, there were two python virtual environments created, one for python exe at the esp tools install path, and another one at my user profile directory. I dont know why and how the user profile virtualenv was created, but it got picked up by VS Code. But after fixing the python paths in the ESP extension's config it works properly now.

BTW, if you could help me with 1 more thing - is there some standard way of handling openocd launch together with xtensa GDB debugger session? I dont see a task for launching gdb in the tasks delivered by esp-idf extension and probably it's because i'm supposed to run the debugger in some different way. But so far i havent got it to work with the built-in gdb or cppgdb task.

brianignacio5, 2020-03-31:
You have a couple of options here:

a) You could use our debug adapter server for Visual Studio Code, it will launch openOCD and GDB based on the extension configuration.
b) You can configure your project's launch json to launch Xtensa GDB and have a preLaunchTask to run openOCD there (or can launch openOCD from the extension commands or lower right start OpenOCD button before start debug).

More info about [debugging configuration](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/DEBUGGING.md)

lafar6502, 2020-03-31:
ok, thank you. still not there with debugging, but we can close this issue now as solved.

closed 2020-03-31 by lafar6502

brianignacio5, 2020-04-01:
You can always post an issue if you find a bug or feature request or post in [Spectrum channel](https://spectrum.chat/espidf-vsc?tab=posts)

artem152012, 2021-02-08:
> You have a couple of options here:
> 
> a) You could use our debug adapter server for Visual Studio Code, it will launch openOCD and GDB based on the extension configuration just running F5.
> b) You can configure your project's launch json to launch Xtensa GDB using the Microsoft C/C++ extension and have a `preLaunchTask: openocd` there to run openOCD there (or can launch openOCD from the extension commands or lower right start OpenOCD button before start debug).
> 
> More info about [debugging configuration](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/DEBUGGING.md)

oh, i havbe the same problem) is it okay?

[image]

brianignacio5, 2021-02-10:
@artem152012 it looks ok, just make sure python packages are installed in this virtual environment as specified in the setup documentation