espressif/vscode-esp-idf-extension #77: Python binary path is not defined

ArTyGill, 2020-04-02:
When the monitor starts, an error pops up, but the monitor seems to be working fine.
[image]

SoumeshBanerjee, 2020-04-02:
Looks like Python binary path is not set properly, please follow the [config guide](https://github.com/espressif/vscode-esp-idf-extension#idf-specific-settings) to set correct Python Path

ArTyGill, 2020-04-02:
@SoumeshBanerjee 
I tried to specify virtual_env and the original python path, it still gives an error when starting the monitor. But as I said, the monitor still works well.

ArTyGill, 2020-04-02:
@SoumeshBanerjee 
Also did configuration several times, all were successful.

ArTyGill, 2020-04-02:
@SoumeshBanerjee But this line and the number of arguments are very confused when executing the Executing monitor.
Executing "C:\Users\artyo\Documents\develop\espressif\.espressif\python_env\idf4.0_py3.8_env\Scripts\python.exe c:\Users\artyo\Documents\develop\espressif\esp-idf\tools/idf_monitor.py -p COM5 -b 115200 c:\users\artyo\documents\work\espressif\oppeneidf\build\OppenEIDF.elf -m 'C:\Users\artyo\Documents\develop\espressif\.espressif\python_env\idf4.0_py3.8_env\Scripts\python.exe' 'c:\Users\artyo\Documents\develop\espressif\esp-idf\tools\idf.py' '-p' 'COM5'"...      
--- idf_monitor on COM5 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

SoumeshBanerjee, 2020-04-02:
> But this line and the number of arguments are very confused when executing the Executing monitor

This is generated by `idf.py monitor` even from cmd and nothing specific to the vscode

ArTyGill, 2020-04-02:
@SoumeshBanerjee
 Ok, then what should I do?

SoumeshBanerjee, 2020-04-02:
> But as I said, the monitor still works well

For further investigation please attach the [logs](https://github.com/espressif/vscode-esp-idf-extension/blob/master/.github/ISSUE_TEMPLATE/bug_report.md)

ArTyGill, 2020-04-02:
OS: Windows 10 1909
IDF-PATH: release/4.0 (master)
VScode: version 1.43.2

Description: 

Error every time you start the monitor.

Log file:
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/4421226/esp_idf_vsc_ext.log)

[image]

SoumeshBanerjee, 2020-04-02:
Seems like either you don't have proper permission for the python path you're setting or the path is incorrect

ArTyGill, 2020-04-02:
@SoumeshBanerjee 
idf.pythonBinPath is not defined.
 
where can i override idf.pythonBinPath?

SoumeshBanerjee, 2020-04-02:
You can do so the extension settings

brianignacio5, 2020-04-03:
Did you follow the [Configure ESP-IDF extension](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/ONBOARDING.md) command?

meltdown03, 2020-04-04:
I get the same error in linux when using the extensions builtin "Monitor" button on the bottom-left. It will still work until I close the Terminal, then I just get the error. The other commands work without the issue. I set up using the Onboarding mutiple times but always get the error. 

I included my log.
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/4431501/esp_idf_vsc_ext.log)

meltdown03, 2020-04-04:
I changed utils.ts to only check that the file exists:
```
export function canAccessFile(filePath: string, mode?: number): boolean {
  try {
    // tslint:disable-next-line: no-bitwise
    // mode = mode || fs.constants.R_OK | fs.constants.W_OK | fs.constants.X_OK;
    fs.accessSync(filePath, fs.constants.F_OK);
    return true;
  } catch (error) {
    Logger.error(`Cannot access filePath: ${filePath}`, error);
    return false;
  }
}
```
It works now without getting the error but still only if I don't close the terminal. Once I do, and try to reopen the monitor, the button does nothing - no error, no monitor, nothing logged.

ArTyGill, 2020-04-11:
Added the variable "idf.pythonBinPathWin" to the settings, now it works without errors. Now we need to understand why pythonPath was not automatically assigned.

[image]

ArTyGill, 2020-04-11:
@meltdown03 
What is your operating system?

meltdown03, 2020-04-11:
Linux, I also replied to the new issue that you created.

SoumeshBanerjee, 2020-04-13:
@ArTyGill We found the root cause and are fixing the issue, once done will tag the commit here

ArTyGill, 2020-04-13:
@SoumeshBanerjee Okay. Thanks!

PR #82 (2020-04-13): IDF Monitor Terminal Issue (VSC-285)
- Fix the Monitor Terminal Restart issue (Closes #81)
- Fix the Python Path not accessible issue (Closes #77)
commit ce552e4e (2020-04-13): :bug: Fix issues with monitor terminal (#82)

- Terminal Restart Failure
- Terminal python path not accessible error

Closes #77 #81
closed 2020-04-13 by SoumeshBanerjee