espressif/vscode-esp-idf-extension #81: When you try to restart the monitor, nothing happens.

ArTyGill, 2020-04-11:
When you try to restart the monitor, nothing happens.

Workaround to run the IDF build, flash, monitor command.

ArTyGill, 2020-04-11:
OS: Windows 10 1909
IDF-PATH: release/4.0 (master)
VScode: version 1.44.0

meltdown03, 2020-04-11:
I have this issue also. Nothing is logged, no error is displayed. It only happens if I close the initial terminal that is opened on the first Monitor launch.

OS: Archlinux
ESP-IDF: master branch 4.2
All packages up-to-date

SoumeshBanerjee, 2020-04-13:
> When you try to restart the monitor, nothing happens.

How do you restart the monitor, is it just launching the monitor clicking the monitor icon at the bottom and then closing the monitor terminal and then launching monitor again by clicking the button ?

brianignacio5, 2020-04-13:
In the log it seems you have  `idf.pythonBinPath is not defined` Do you configure the extension using `ESP-IDF: Configure extension` command ?

meltdown03, 2020-04-13:
That's what I do. I also press ctrl-] to stop the monitor before closing the terminal. The tasks all work everytime, so do the other icons, just not the monitor icon.

SoumeshBanerjee, 2020-04-13:
@meltdown03 Thanks for the feedback, we found the issue and working to fix the same

SoumeshBanerjee, 2020-04-13:
Possible duplicate of #77

ArTyGill, 2020-04-13:
@SoumeshBanerjee Yes, the same problem, the only option that works well is the "build, flash, monitor" button, which displays the message "idf monitor was closed" before starting a new task.

PR #82 (2020-04-13): IDF Monitor Terminal Issue (VSC-285)
- Fix the Monitor Terminal Restart issue (Closes #81)
- Fix the Python Path not accessible issue (Closes #77)
closed 2020-04-13 by SoumeshBanerjee