espressif/vscode-esp-idf-extension #83: Hexadecimal menuconfig items are saved in decimal

boarchuz, 2020-04-15:
After changing a menuconfig item with a hex value and clicking the save button, its "0x" prefix is truncated in the updated sdkconfig file.

Taking CONFIG_USB_DESC_CUSTOM_VID, for example, with this default value in sdkconfig:

> CONFIG_USB_DESC_CUSTOM_VID=0x1234

 After 'Launch GUI configuration tool' -> changing this value to 0x1235 -> Save:

> CONFIG_USB_DESC_CUSTOM_VID=1235

Interestingly, it appears correctly as "0x1235" in the GUI, even after reopening Code.

Extension 0.2.0

brianignacio5, 2020-04-16:
With @projectgus we implemented a small fix in ESP-IDF 's confserver for current master for the returned value.

Any thoughts @projectgus ?

brianignacio5, 2020-04-16:
Because in the GUI menuconfig we have a mask to pre append 0x to the value.

The confserver from ESP-IDF still returns the hex number (without 0x) and saves it like so. 

As I understand, Kconfiglib identify this config as hex so it threats the number as hex anyways. Any thoughts @projectgus ?

boarchuz, 2020-04-16:
I should have noted that I first hit this issue after altering the partition table offset. 
In that case, flash failed because write_flash was being called with "8000" for partition-table.bin rather than "0x8000", with an error message indicating a conflict at 0x1f40 (ie. it was in fact being interpreted as a decimal value).

projectgus, 2020-04-16:
Hi @boarchuz @brianignacio5 ,

It looks like this is a bug in IDF's cmake generation from sdkconfig. Without the hex prefix the contents of `build/sdkconfig.h` and `build/sdkconfig.json` are correct, but `sdkconfig.cmake` has the decimal value.

We'll fix this on the ESP-IDF side so the hex value is correctly interpreted even if it's missing the `0x` prefix.

closed 2020-05-14 by github-actions[bot]

brianignacio5, 2020-05-14:
Hi @boarchuz  this issue is currently being fixed in ESP-IDF and will be released in master soon.

Will update here when it does.