espressif/vscode-esp-idf-extension #84: VScode + Extension show "sizeof(void *)" is 8UL.

pmnxis, 2020-04-18:
My vscode and esp-idf extension show sizeof pointer is 8 byte length.
As I know, xtensa lx6 architecture is 32bit address system and this show me strangely.

This is not critical issue. But I thought we need to check this issue surely. So update issue here. Thanks

Screenshots
[image]

Environment

- OS Name & Version: MacOS X 10.15.4
- VSCode Version: 1.44.1
- ESP-IDF Version: 4.0.0
- Python Version: 2.7.16

SoumeshBanerjee, 2020-04-20:
@pmnxis C/C++ intellisense is provided by a separate `C/C++ Extension` which you can [configure here](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/C_CPP_CONFIGURATION.md) and this extension is not maintained by us, but that being said we are constantly working to improve the experience for our extension

Once we find improvements to the intellisense, will let you know.

boarchuz, 2020-04-22:
Can you try opening c_cpp_properties.json and changing intelliSenseMode to "gcc-x86"? 

Note that changing this can be very 'sticky' in my experience- even after restarting VS Code or resetting its database, it may still show 8UL. What works reliably to force a cpp refresh is to change to a different cpp configuration and then back again (of course you'll need >1 configuration in c_cpp_properties.json to do this).

closed 2020-05-13 by github-actions[bot]