espressif/vscode-esp-idf-extension #89: IDF_PATH environment variable is different from inferred IDF_PATH

edouardreg, 2020-04-24:
Hello,

**Describe the bug**
I get the warning from CMAKE : IDF_PATH environment variable is different from inferred IDF_PATH

**To Reproduce**
Build the template created by the VScode xtension

**Expected behavior**
Build without warning

**Environment (please complete the following information):**

- OS Name & Version: latest windows 10 pro 64 bits. Windows_NT x64 10.0.18363
- VSCode Version: Version: 1.44.2
- ESP-IDF Version: v4.0
- Python Version: 3.8

**Logs**
If applicable, please share the log file which can be obtained from

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log : can't share the log (file not supported ?)

**Additional context**
build output : 

```
⚠️ CMake Warning at H:/Espressif/esp-idf/tools/cmake/idf.cmake:14 (message):
  IDF_PATH environment variable is different from inferred IDF_PATH.

                              Check if your project's top-level CMakeLists.txt includes the right
                              CMake files. Environment IDF_PATH will be used for the build.
Call Stack (most recent call first):
  H:/Espressif/esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:6 (include)

```

edouardreg, 2020-04-24:
Try to share log zipped :

[esp_idf_vsc_ext.zip](https://github.com/espressif/vscode-esp-idf-extension/files/4527361/esp_idf_vsc_ext.zip)

SoumeshBanerjee, 2020-05-05:
@edouardreg is the build failing as well?

brianignacio5, 2020-05-09:
Well the main reason this happened is because we were not assigning the python env variable that Build system use nowadays (PYTHON), this should be fixed in GitHub master here:

https://github.com/espressif/vscode-esp-idf-extension/blob/master/src/utils.ts#L579

edouardreg, 2020-05-09:
> @edouardreg is the build failing as well?

No, the build is ok.

edouardreg, 2020-05-09:
> Oh yes the reason for this is that in your project's top CMakeLists.txt you have ENV{IDF_PATH} which get your system IDF_PATH instead of the one used in the build child process environment, which comes from the extension `idf.espIdfPath`. This mean you are working with another IDF_PATH that is not the one in your environment variables.
> 
> 1. You can ignore the warning since it just means exactly what it says. Build should work.
> 2. You can set your extension `idf.espIdfPath` equal to your system env variable IDF_PATH .

Ok : thanks I will try on Monday.

mitchjs, 2020-05-11:
i get this error too, and cannot build from the status bar icon anymore...
fails with this error: Make sure you have the build tools installed and set in $PATH

edouardreg, 2020-05-15:
So I set the idf.espIdfPath identically as my env variable : but still the same warning.
So I just ignore it (build is ok).

closed 2020-06-09 by github-actions[bot]