espressif/vscode-esp-idf-extension #91: Error In Launch Gui Configuration Tool

sunil19822701, 2020-04-29:
I get the following error when launching gui configuration tool
---------------------------ERROR--------------------------

Error: spawn python ENOENT
	at Process.ChildProcess._handle.onexit (internal/child_process.js:264:19)
	at onErrorNT (internal/child_process.js:456:16)
	at processTicksAndRejections (internal/process/task_queues.js:77:11)
-----------------------END OF ERROR-----------------------

brianignacio5, 2020-04-30:
Please take a look at https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/ONBOARDING.md to properly configure your extension.

sunil19822701, 2020-04-30:
I have configured the extension as per the document and i am able to build, monitor and flash it on esp32 but not able to launch gui configuration tool from vscode.

brianignacio5, 2020-04-30:
Can you provide more information about your setup? 

**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [ ] Make sure you have searched for existing bugs and features request before you post an issue.
- [ ] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [ ] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
A clear and concise description of what the bug is (current behavior).

**To Reproduce**
Steps to reproduce the behavior (including the full log).

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

- OS Name & Version: [e.g. MacOS X 10.14.5]
- VSCode Version: [e.g. 1.35.1]
- ESP-IDF Version: [e.g. v3.2.2]
- Python Version: [e.g. 2.7.10]

**Logs**
If applicable, please share the log file which can be obtained from

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

**Additional context**
Add any other context about the problem here.

sunil19822701, 2020-04-30:
ok solved 
i had to replaced all the python bin path with absolute bin path as it was set to "python" in extension settings.
After Onboarding Complete only absolute path was set for idf: Python System Bin Path Win

closed 2020-04-30 by sunil19822701