espressif/vscode-esp-idf-extension #93: Error on build in version 0.2.1

mvacha, 2020-05-02:
**Describe the bug**
Build randomly fails with error: "Make sure you have the build tools installed and set in $PATH". 
After a restart it usually works, but not always. When it fails, there is no message in "ESP-IDF Build" output. The behavior is the same whether esp-idf is installed using the Windows installer or the wizard in the extension. 

Downgrading the extension to 0.2.0 solves the problem.

**To Reproduce**
1. Install ESP-IDF using Windows installer or using the wizard in the vscode-esp-idf extension.
1. Run either "ESP-IDF: Build your project" task or click the build button in status bar.

**Expected behavior**
Project gets build as ESP-IDF is installed and the extension is correctly configured.

**Environment (please complete the following information):**

- OS Name & Version: Windows 10 1909
- VSCode Version: 1.44.2
- ESP-IDF Version: 4.0.0
- Python Version: 3.8

**Logs**
```
{"user":true,"message":"Make sure you have the build tools installed and set in $PATH","stack":"Error: spawn cmake ENOENT\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:264:19)\n\tat onErrorNT (internal/child_process.js:456:16)\n\tat processTicksAndRejections (internal/process/task_queues.js:77:11)","level":"error","timestamp":"2020-05-02T20:03:51.383Z"}
```

jmpmscorp, 2020-05-04:
Hi.

I have noticed same issue. In my case, error only is shown if I open GUI configuration tool. Everytime I open it, change any parameter and click on save, when I try to build and upload firmware with new changes, @mvacha error is shown too.

If I restart vscode, upload works again without problem. However, if I open GUI configuration tool again, error appears again, so it seems problem is related with GUI tool.

Regards.

SoumeshBanerjee, 2020-05-05:
@jmpmscorp are you also facing the problem on a windows machine ?

jmpmscorp, 2020-05-05:
Hi @SoumeshBanerjee.

Yes, my machine has Windows 10 version. If you need, I can test it with an Ubuntu 20.04 virtual machine I have mounted with vscode and different esp-idf framework version.

SoumeshBanerjee, 2020-05-05:
> If you need, I can test it with an Ubuntu 20.04

Sure, Thanks

jmpmscorp, 2020-05-05:
Hi again @SoumeshBanerjee .

Tested with Ubuntu 20.04. No error is shown.

SoumeshBanerjee, 2020-05-05:
@jmpmscorp Thanks 

> Tested with Ubuntu 20.04. No error is shown.

Same for my case no error on Mac and Linux; once I have access to the windows machine/vm will test this problem and let you guys know

Thanks for reporting and helping

MervinKoops, 2020-05-06:
Having the same issue after the ESP-IDF extension updated.
Tried configuration in the 'onboarding' page again with no change. re-adding .vscode folder via extension doesn't work either.
- OS: Windows 10, v1909
- VSCode: 1.43.2
- ESP-IDF: v4
- Python: 3.8

brianignacio5, 2020-05-08:
After you install the extension and run the onboarding page this error happens. Have anyone tried closing vscode and reopen it and see if they can build ? 
@MervinKoops @mvacha @jmpmscorp ?

mvacha, 2020-05-08:
@brianignacio5 yes, this usually fixed the issue, but it was not a reliable "fix". But opening a second instance of VS Code would trigger this bug again. After downgrading to 0.20 I've never experienced this bug again.

brianignacio5, 2020-05-09:
I only managed to reproduce on Win 10 VM on first install, probably buildManager is not getting the new values right away but after vscode restart it works ok. Haven't been able to reproduce this for example on multiple vscode windows opened.

Could you please check if there was any error during IDF Onboarding and the values in `idf.customExtraPaths` and `idf.espIdfPath` are correct ? Also consider that if your project has `.vscode/settings.json` they will override values from workspace settings and user settings.

mitchjs, 2020-05-11:
i got this error now.. its all the time i cannot build from the build ICON on the bottom bar anymore
i dont think #95 does anything to help

no errors during onboarding...

brianignacio5, 2020-05-11:
Have you tried #95 @mitchjs ?

MervinKoops, 2020-05-11:
I've tried a restart of VSCode and Windows, no changes.

When looking at "idf.customExtraPaths" and "idf.espIdfPath" I see a mixed usage of "\" (backslash) and "\\" double backslash and some single slashes "/".  What would be the correct thing?

mvacha, 2020-05-11:
@brianignacio5 I did more testing:
Upgrading from working 0.20 to 0.21 breaks the build and no amount of VSCode restarts seems to fix it.
These are my idf settings:
```
    "idf.toolsPathWin": "C:\\Users\\michal\\.espressif",
    "idf.pythonBinPathWin": "C:\\Users\\michal\\.espressif\\python_env\\idf4.0_py3.8_env\\Scripts\\python.exe",
    "idf.customExtraVars": "{\"IDF_CCACHE_ENABLE\":\"1\",\"OPENOCD_SCRIPTS\":\"C:\\\\Users\\\\michal\\\\.espressif\\\\tools\\\\openocd-esp32\\\\v0.10.0-esp32-20190313/openocd-esp32/share/openocd/scripts\"}",
    "idf.espIdfPathWin": "c:\\Users\\michal\\esp-idf",
    "idf.customExtraPaths": "C:\\Users\\michal\\.espressif\\tools\\xtensa-esp32-elf\\esp-2019r2-8.2.0\\xtensa-esp32-elf\\bin;C:\\Users\\michal\\.espressif\\tools\\esp32ulp-elf\\2.28.51.20170517\\esp32ulp-elf-binutils\\bin;C:\\Users\\michal\\.espressif\\tools\\cmake\\3.13.4\\bin;C:\\Users\\michal\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20190313\\openocd-esp32\\bin;C:\\Users\\michal\\.espressif\\tools\\mconf\\v4.6.0.0-idf-20190628\\;C:\\Users\\michal\\.espressif\\tools\\ninja\\1.9.0\\;C:\\Users\\michal\\.espressif\\tools\\idf-exe\\1.0.1\\;C:\\Users\\michal\\.espressif\\tools\\ccache\\3.7\\;C:\\Users\\michal\\.espressif\\python_env\\idf4.0_py3.8_env\\Scripts;C:\\Users\\michal\\esp-idf\\tools\t",
```

After running the Onboarding again, there is no error in the output, the aforementioned idf paths are the same and build is still broken for me no matter how many times I restart the VS Code.
```
idf_versions.txt has (1) KB
idf_versions.txt progress: 100.00% (0.04 / 0.04) KB
Checking ESP-IDF Python requirements using C:\Users\michal\.espressif\python_env\idf4.0_py3.8_env\Scripts\python.exe
Python requirements from c:\Users\michal\esp-idf\requirements.txt are satisfied.

Checking Debug Adapter requirements using C:\Users\michal\.espressif\python_env\idf4.0_py3.8_env\Scripts\python.exe
Python requirements from c:\Users\michal\.vscode\extensions\espressif.esp-idf-extension-0.2.1\esp_debug_adapter\requirements.txt are satisfied.

The following paths would be added to env PATH
C:\Users\michal\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin;C:\Users\michal\.espressif\tools\esp32ulp-elf\2.28.51.20170517\esp32ulp-elf-binutils\bin;C:\Users\michal\.espressif\tools\cmake\3.13.4\bin;C:\Users\michal\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190313\openocd-esp32\bin;C:\Users\michal\.espressif\tools\mconf\v4.6.0.0-idf-20190628\;C:\Users\michal\.espressif\tools\ninja\1.9.0\;C:\Users\michal\.espressif\tools\idf-exe\1.0.1\;C:\Users\michal\.espressif\tools\ccache\3.7\;C:\Users\michal\.espressif\python_env\idf4.0_py3.8_env\Scripts;C:\Users\michal\esp-idf\tools	
```

I've tried #95, but I get the exact same error.

jmpmscorp, 2020-05-11:
Hi @brianignacio5 

Excuse no response. I have been busy last days.

> After you install the extension and run the onboarding page this error happens. Have anyone tried closing vscode and reopen it and see if they can build ?

As I wrote before, everytime I restarted vscode, build worked again, but If I opened GUI tool, build fialed until new restart.

Now, I have another issue, I think is related with this. This weekend I had to clean and format my Windows PC. After fresh OS, IDF tools and vscode + idf plugin installation, I'm facing new problem.

Before 0.21 installation, onboarding configuration success, I can't start GUI tool when build folder is present. I open project and click on GUI tool button and this error is shown:

```
[0/1] cmd.exe /C "cd /D F:\JM\Code\esp-idf\sht31
_test\build && python F:/Tools/espressif/esp-idf-4.1/tools/kconfig_new/prepare_kconfig_files.py --env-file F:/JM/Code/esp-idf/sht31_test/build/config.env && python F:/Tools/espressif/esp-idf-4.1/tools/kconfig_new/confserver.py --env-file F:/JM/Code/esp-idf/sht31_test/build/config.env --kconfig F:/Tools/espressif/esp-idf-4.1/Kconfig --sdkconfig-rename F:/Tools/espressif/esp-idf-4.1/sdkconfig.rename --config F:/JM/Code/esp-idf/sht31_test/sdkconfig"

---------------------------ERROR--------------------------

"python" no se reconoce como un comando interno o externo,
programa o archivo por lotes ejecutable.

-----------------------END OF ERROR-----------------------
FAILED: CMakeFiles/confserver 
cmd.exe /C "cd /D F:\JM\Code\esp-idf\sht31_test\build && python F:/Tools/espressif/esp-idf-4.1/tools/kconfig_new/prepare_kconfig_files.py --env-file F:/JM/Code/esp-idf/sht31_test/build/config.env && python F:/Tools/espressif/esp-idf-4.1/tools/kconfig_new/confserver.py --env-file F:/JM/Code/esp-idf/sht31_test/build/config.env --kconfig F:/Tools/espressif/esp-idf-4.1/Kconfig --sdkconfig-rename F:/Tools/espressif/esp-idf-4.1/sdkconfig.rename --config F:/JM/Code/esp-idf/sht31_test/sdkconfig"
ninja: build stopped: subcommand failed.

Executing action: confserver
Running ninja in directory f:\jm\code\esp-idf\sht31_test\build
Executing "ninja confserver"...
ninja failed with exit code 1

---------------------------ERROR--------------------------

Received signal: null with code: 2
-----------------------END OF ERROR-----------------------
```
Howerver, if I delete build folder, restart vscode and click again on GUI button, no error is shown and GUI tool opens and works great.

Now, I click on build button. Build process complete succesfully and I can flash without issue. Restart again vscode, click on GUI tool button to change any parameter and same error again appears, so GUI tool doesn't work again.

I can check that everytime I delete build folder, GUI tool starts and works, but when I build project, it crash again and again.

I have downgraded to 0.20 version and everything works perfectly. Any error is shown.

How can I check #95? Compiling from source?

mitchjs, 2020-05-11:
i tried PR #95 and no change... same error when i try to build from the build icon on bottom bar

mitchjs, 2020-05-11:
while on a path topic

i was just looking at settings... and see

{"IDF_CCACHE_ENABLE":"1","OPENOCD_SCRIPTS":"C:\\Users\\mitchjs\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20190313/openocd-esp32/share/openocd/scripts"}

there is a mix of path delimiters \\ and /
is that correct?

brianignacio5, 2020-05-12:
The reason for this is that OPENOCD_SCRIPTS is generated using `tools.json` from ESP-IDF (or from the extension as fallback) which has hard-coded the last part. We could probably enhance this later since openocd server is working from the extension.

```
OPENOCD_SCRIPTS": "${TOOL_PATH}/openocd-esp32/share/openocd/scripts"
```

brianignacio5, 2020-05-12:
Please try the latest commit @mvacha It seems to be working for @mitchjs .

You can clone this repository, pick branch of #95 and follow the steps in https://github.com/espressif/vscode-esp-idf-extension#build-from-source-code @jmpmscorp

mvacha, 2020-05-12:
@brianignacio5 I've just tried it and it fixes the issue for me :) Thx for fixing it.

jmpmscorp, 2020-05-12:
@brianignacio5 I have tried #95 too. It seems to work.

However, I have noticed strange behaviour. Starting with a fresh project (without build folder):

1. Build click, works, Open GUI works. Restart vscode. Open GUI tools, works, build works.
2. Open GUI works, build works. Restart vscode. Open GUI fails, build works. Restart vscode. GUI fails, build works. Every restart same problem happens.

So, I have noticed that in order to use GUI tools without problems, you need to build firmware previously. 

Everytime I deleted build folder and followed process 1, GUI and build works. On the other hand, process 2 always failed.

mitchjs, 2020-05-12:
@jmpmscorp 

explain what to do for the fails, and ill try to reproduce on my system
im running v0.2.1 with pr #95 on win 10

jmpmscorp, 2020-05-12:
@mitchjs 

Please, create new project, so no build folder exist, and follow point 2.

Open GUI tools -> Build project -> Restart VSCode -> Open GUI tools -> Build project.

Following this process and after vscode restart, when I try to open GUI tools I received this error in console:

```
[0/1] cmd.exe /C "cd /D F:\JM\Code\esp-idf\sht31
_test\build && python F:/Tools/espressif/esp-idf-4.1/tools/kconfig_new/prepare_kconfig_files.py --env-file F:/JM/Code/esp-idf/sht31_test/build/config.env && python F:/Tools/espressif/esp-idf-4.1/tools/kconfig_new/confserver.py --env-file F:/JM/Code/esp-idf/sht31_test/build/config.env --kconfig F:/Tools/espressif/esp-idf-4.1/Kconfig --sdkconfig-rename F:/Tools/espressif/esp-idf-4.1/sdkconfig.rename --config F:/JM/Code/esp-idf/sht31_test/sdkconfig"

---------------------------ERROR--------------------------

"python" no se reconoce como un comando interno o externo,
programa o archivo por lotes ejecutable.

-----------------------END OF ERROR-----------------------
FAILED: CMakeFiles/confserver 
cmd.exe /C "cd /D F:\JM\Code\esp-idf\sht31_test\build && python F:/Tools/espressif/esp-idf-4.1/tools/kconfig_new/prepare_kconfig_files.py --env-file F:/JM/Code/esp-idf/sht31_test/build/config.env && python F:/Tools/espressif/esp-idf-4.1/tools/kconfig_new/confserver.py --env-file F:/JM/Code/esp-idf/sht31_test/build/config.env --kconfig F:/Tools/espressif/esp-idf-4.1/Kconfig --sdkconfig-rename F:/Tools/espressif/esp-idf-4.1/sdkconfig.rename --config F:/JM/Code/esp-idf/sht31_test/sdkconfig"
ninja: build stopped: subcommand failed.

Executing action: confserver
Running ninja in directory f:\jm\code\esp-idf\sht31_test\build
Executing "ninja confserver"...
ninja failed with exit code 1

---------------------------ERROR--------------------------

Received signal: null with code: 2
-----------------------END OF ERROR-----------------------
```

But If process is (from fresh project or deleted build folder): Build project -> Open GUI ->Restart vscode ->Open GUI -> Build project, any error is shown and everything work as expected.

brianignacio5, 2020-05-13:
@jmpmscorp it seems that when you start the guiconfig the value in `idf.pythonBinPath` is not updated from its default value. 

Did you configure your extension using the command `ESP-IDF: Configure ESP-IDF extension` for the user/workspace settings? Please make sure your current project if `idf.pythonBinPath`

What is the value of `idf.pythonBinPath`

jmpmscorp, 2020-05-13:
@brianignacio5 I'm becoming crazy with GUI and 0.21 xD. I can't figure out how to resolve it. I have completed onboarding configuration around 20 times, checking paths but I can't find where the problem is. If I downgrade to 0.20 everything works perfectly.

I attach many screenshots I have done. Feel free to delete if there are too many pictures.

Onboarding python:
[image]

Onboarding tools:
[image]

vscode esp-idf settings where I have found "python related config":
[image]
[image]

My python env PATH:
[image]
[image]

mitchjs, 2020-05-13:
@jmpmscorp did you use v0.2.1 and merged with #95?
i can give you the visx i created from ver0.2.1 and #95

jmpmscorp, 2020-05-13:
OMG... I finally have it... Thanks @mitchjs.

I was compiled and installed master branch, I think, without #95. I have downloaded directly @brianignacio5 bugfix/env-path branch. After compile it, packaged in vsix and installed, now, I haven't any problem with build and GUI.

Thank you both to help me, :)

mitchjs, 2020-05-13:
@jmpmscorp... woohoo!

brianignacio5, 2020-05-14:
Sorry about this @jmpmscorp Will merge to master soon.

We always trying to improve the extension, we are working on adding cool features like gcov coverage, ESP-IDF rainmaker integration, heap tracing and more !

PR #95 (2020-05-18): fix env path in windows and python on terminal (VSC-319)
Closes #93, #100 

Normalize modifiedEnv PATH and Path updating. 

Also fix wrong python picked from ESP-IDF Terminal (should use the virtualenv)
closed 2020-05-18 by brianignacio5