espressif/vscode-esp-idf-extension #96: Completely lost with errors setting up VSCode ESP-IDF extension

yankee14, 2020-05-13:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [ ] Make sure you have searched for existing bugs and features request before you post an issue.
- [ ] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [ ] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
I made a video showing ESP-IDF working from the terminal, followed by the VSCode issue: https://www.youtube.com/watch?v=WJATDQhhZM4&feature=youtu.be

ESP-IDF works correctly on the command line after following instructions from Espressif, but I'm having issues understanding what the issue is with the VSCode extension. I hope it's just a missing configuration option that I overlooked, but I'm not sure yet.

- Intellisense is not understanding some of the ESP-IDF library variables, such as `portTICK_PERIOD_MS`.

I have encountered multiple errors in the terminal section of VSCode when trying to build, including:
- `identifier "CONFIG_FREERTOS_HZ" is undefined [35,27]`
- `The task provider for "C/C++" tasks unexpectedly provided a task of type "shell".
`
- `> Executing task: /usr/bin/python /home/yankee/build/src/esp-idf/tools/idf.py build <
'cmake' must be available on the PATH to use /home/yankee/build/src/esp-idf/tools/idf.py
The terminal process terminated with exit code: 2
Terminal will be reused by tasks, press any key to close it.`

**To Reproduce**
1. Setup ESP-IDF via the command line and show that it is working.
2. Install VSCode and the ESP-IDF extension
3. Setup ESP-IDF extension following onboarding instructions, using existing ESP-IDF from step 1.
4. Look at extra instructions from VSCode Extension GitHub page
5. Try to run `hello_world` ESP32 project
6. Encounter many errors

**Expected behavior**
Once the onboarding process confirmed the existence and location of the build dependencies, I expected it to be able to build without an issue.

**Screenshots**
https://www.youtube.com/watch?v=WJATDQhhZM4&feature=youtu.be

**Environment (please complete the following information):**

- OS Name & Version:

```
$ uname -a
Linux ibn5100 5.6.0-1-amd64 #1 SMP Debian 5.6.7-1 (2020-04-29) x86_64 GNU/Linux

$ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux bullseye/sid"
NAME="Debian GNU/Linux"
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

- VSCode Version:

```
$ code --version
1.45.0
d69a79b73808559a91206d73d7717ff5f798f23c
x64
```

- ESP-IDF Version:

```
$ git log
commit b0f053d82dc4484479521a0d49a0b6917946602a (HEAD -> release/v4.0, origin/release/v4.0)
```

- Python Version:

```
$ python --version
Python 3.8.3rc1
```

**Logs**
If applicable, please share the log file which can be obtained from

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/4619458/esp_idf_vsc_ext.log)

**Additional context**
Add any other context about the problem here.

SoumeshBanerjee, 2020-05-13:
For building directly with `idf.py build` it seems like you're using `make` build system while our extension only supports `cmake` build, please see [Prerequisite](https://github.com/espressif/vscode-esp-idf-extension#prerequisites)

Also the launch which you used using <kbd>F5</kbd>, will open debugger and more info on debugging can be [found here](https://github.com/espressif/vscode-esp-idf-extension#debugging)

Also follow the steps here in [onboarding flow to setup](https://github.com/espressif/vscode-esp-idf-extension#esp-idf-configure-extension)

And for building we recommend using either build command directly from command palate (_which can be accessed by <kbd>F1</kbd>_) or using the little cylinder icon at the bottom tray in vscode. Also please take a look at the list of the commands [here](https://github.com/espressif/vscode-esp-idf-extension#available-commands)

yankee14, 2020-05-13:
@SoumeshBanerjee hi. I am not sure that `idf.py build` automatically implies a legacy build instead of a cmake build. If you look at the instructions here, you can see that Espressif made idf.py a wrapper for cmake after version 4: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html

I am not trying to setup the debugger yet. That's after I get it to build.

I will try your other instructions tomorrow. Thanks for your help!

SoumeshBanerjee, 2020-05-13:
> If you look at the instructions here, you can see that Espressif made idf.py a wrapper for cmake after version 4

Probably you ask folks at `espressif/esp-idf`, for raising IDF related bugs please [visit here](https://github.com/espressif/esp-idf/issues)

yankee14, 2020-05-13:
I don't think this is an issue with Espressif's tools, though. I show that their tools work fine in the video.

brianignacio5, 2020-05-13:
The main issue with your setup is that you are skipping the Download Tools step which does:
1) Download the tools and create the virtual env for python and install dependencies there.
2) Set the virtual python full path in `idf.pythonBinPath` which as explained in onboarding document is the python executable used in the extension with fallback in `python`.
3) Set `idf.customExtraVars` to correct path. On the skip download tools, you didn't replace `${TOOL_PATH}` of `${TOOL_PATH}/openocd-esp32/share/openocd/scripts` for OPENOCD_SCRIPTS which is why openocd is probably failing.
4) Set `idf.customExtraPaths` but you can manually do so in the Skip Download Tools section. The thing is that the Download section also appends the python executable parent folder to allow the extension commands and tasks to obtain python (think as it as a env PATH replacement). 

The reason that the build task fails is because in the template tasks, there is an option **env** with is using PATH = idf.customExtraPaths (doesn't override system PATH). 

We can't update ${TOOL_PATH} in the Skip Sections if we don't know where openocd is. We could make a trigger based on the paths input, but that would complicate the check tools action.

I'm sorry for this situation, we are working to try to automate/improve as much of this as possible.

There is a PR that has the `idf.pythonBinPath` input in the Skip Download Tools section which makes it more sense, but is part of a bigger feature (allow the user to change esp-idf versions and IDF Tools based on previously downloaded versions). An smaller PR with this feature should makes it easier.

Some reasons we try to isolate the PATH used in the extension from system PATH:
1. Conflict with other extensions. Some extension modifies the environment variables and that makes our extension subject to unpredictable changes.
2. We don't to modify your system in any way.
3. We want to be able to support an isolated environment which allows you to have different setups (that's why there user, workspace and project settings option in onboarding)

yankee14, 2020-05-13:
The things you've described roughly make sense to me. Is there a way to trigger the onboarding process whenever I want to, so that I can play with it? I need to be able to see these variables and what they are set to:

[image]

But I can only see them after going through that onboarding process. I don't see them in any of these JSON files:

[image]

Anyways, I experimented with PATH a little bit by appending my normal BASH terminal's path, which looks like this:

```
$ printf -- "${PATH}\n"
/home/yankee/.local/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games:/home/yankee/.local/bin:/home/yankee/.local/bin:/home/yankee/.local/bin
```

So, I just tacked that onto the end of the path in the onboarding process. I got a bit further in the build process, but hit another snag. It seems to not be using my pip environment that has the prerequisites:

```
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
The following Python requirements are not satisfied:
click>=5.0
pyserial>=3.0
future>=0.15.2
pyparsing>=2.0.3,<2.4.0
pyelftools>=0.22
To install the missing packages, please run "/home/yankee/build/src/esp-idf/install.sh"
Alternatively, you can run "/usr/bin/python -m pip install --user -r /home/yankee/build/src/esp-idf/requirements.txt" for resolving the issue.
⚠️ CMake Error at /home/yankee/build/src/esp-idf/tools/cmake/build.cmake:270 (message):
  Some Python dependencies must be installed.  Check above message for
  details.
Call Stack (most recent call first):
  /home/yankee/build/src/esp-idf/tools/cmake/build.cmake:396 (__build_check_python)
  /home/yankee/build/src/esp-idf/tools/cmake/project.cmake:348 (idf_build_process)
  CMakeLists.txt:6 (project)
```

If I could specify the python pip environment to use, it would know those tools are available.

brianignacio5, 2020-05-14:
As explained in the [Readme Env variables section](https://github.com/espressif/vscode-esp-idf-extension#use-of-environment-variables-in-esp-idf-settingsjson-and-tasksjson) replace ${HOME} with ${env:HOME}. 

Why not just use ${HOME} ? Because vscode also use things like ${config:idf.espIdfPath} to refer to extension settings (like in tasks.json) which makes parsing ${.*} requires a bit more context.

yankee14, 2020-05-14:
@brianignacio5 thank you for the help! That makes sense.

brianignacio5, 2020-05-20:
So it seems this issue is fixed for the author. Closing this for now.

Please let us know if this still happens to any of you after v0.2.3 release.

closed 2020-05-20 by brianignacio5