espressif/vscode-esp-idf-extension #100: Make sure you have the build tools installed and set in $PATH

syeedhaiderali, 2020-05-16:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [ ] Make sure you have searched for existing bugs and features request before you post an issue.
- [ ] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [ ] I've read the docs and found no information that could have helped solving the issue.

I had used ESP-IDF with vs code and it worked fine but after a month it stops working and showing error. Kindly guide

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Environment**

- OS Name & Version: [Windows 10]
- VSCode Version: [e.g. 1.45.1]
- ESP-IDF Version: [e.g. v4.0]
- Python Version: [e.g. 3.7.3]

**Logs**
{"user":true,"message":"Make sure you have the build tools installed and set in $PATH","stack":"Error: spawn cmake ENOENT\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:264:19)\n\tat onErrorNT (internal/child_process.js:456:16)\n\tat processTicksAndRejections (internal/process/task_queues.js:77:11)","level":"error","timestamp":"2020-05-16T07:32:17.171Z"}

mitchjs, 2020-05-16:
you need to get PR #95 merged in and build you own vsix and install to VSCode
as of v0.2.1 which is causing this issue
if you dont know or cant set up to build your own vsix
(it was pretty easy)
i can send you a vsix with pr #95 in it

SoumeshBanerjee, 2020-05-17:
Possible duplicate #93

aks11, 2020-05-17:
can someone provide the vsix file with PR#95

mitchjs, 2020-05-18:
@aks11 this should be it... it will install as v0.2.1-pr95
disable auto update of extensions, or you'll get v0.2.1 

[esp-idf-extension.zip](https://github.com/espressif/vscode-esp-idf-extension/files/4641561/esp-idf-extension.zip)

hmcosentini, 2020-05-18:
@mitchjs Thanks I disabled the auto-update and installed from the .vsix it is working. When you plan to release the update in vscode?

SoumeshBanerjee, 2020-05-18:
We have a plan to release this as soon as #95 is merged (possibly today EOD or tomorrow)

syeedhaiderali, 2020-05-18:
Thanks @mitchjs it's working now. @SoumeshBanerjee thanks to you also and I am waiting for the new release.

closed 2020-05-18 by syeedhaiderali