espressif/vscode-esp-idf-extension #105: Configure ESP-IDF extension fails at the stage of downloading tools at Python 3.8

an-dr, 2020-05-18:
**Describe the bug**
Hi guys! I'm preparing a debugging training right now and experimenting with the clean installation of the extension.

I've faced with the bug breaking the configuration process with the Python 3.8.2 (Python 2 works).

**To Reproduce**
- Create a new project
- Call >ESP-IDF: Configure ESP-IDF extension
- At the step of downloading tools you'll get the error

**Expected behavior**
Installing of the tools and appearing of the button "Next"

**Screenshots**
[image]

**Environment (please complete the following information):**

- OS Name & Version: Ubuntu 20.04
- VSCode Version: 1.45.1
- ESP-IDF Version: 4.2
- Python Version: 3.8.2

**Logs**
[esp-idf_output_panel.log](https://github.com/espressif/vscode-esp-idf-extension/files/4643907/esp-idf_output_panel.log)
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/4643908/esp_idf_vsc_ext.log)

brianignacio5, 2020-05-19:
You are missing the python package `bdist_wheel` which is not part of IDF_PATH/requirements.txt or $EXTENSION_DIR/esp_debug_adapter/requirements.txt.

Should we add it to DAP requirements.txt or specify it is needed in the docs?

an-dr, 2020-05-19:
@brianignacio5 `bdist_wheel` is not a standalone package. As I understand, it is a part of either wheel or setuptools. Anyway, I have installed both.
Investigating further

brianignacio5, 2020-05-28:
`pip install --upgrade pip` wheel and `pip3 install --upgrade pip wheel` before configure esp-idf extension ?

ebabcock72, 2020-05-28:
I am having the exact same issue. Your suggested "--upgrade" commands did not work for me.

PR #113 (2020-06-18): :bug: Fix python venv install wheel issue in Linux (VSC-348)
For Python > 3.3, we use `venv` since it the official Python virtual environments module. The thing is, for Linux users it seems to not include `wheel` from the system wide Python within the virtual environment. 

We added an step to install wheel in the virtualenv before installing requirements.txt

Should fix #105
closed 2020-06-18 by brianignacio5