espressif/vscode-esp-idf-extension #106: ESP-IDF Terminal environment issues with IDF_PATH and PowerShell

oakwhiz, 2020-05-20:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [x] Make sure you have searched for existing bugs and features request before you post an issue.
- [x] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [x] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
If the shell is PowerShell (which is the default for VS Code on Windows 10) then `set` actually aliases to `Set-Variable` and creates an empty variable with a very long name of the form `IDF_PATH=C:\Users\user\esp-idf` which is not expected behavior at all.
I discovered this issue when doing `start cmd` in order to get a detached external shell with the same environment, then doing `idf.py menuconfig` to work around other existing issues with menuconfig/kconfig, including: #5 #6 #13 #19 
I believe this bug is located here: https://github.com/espressif/vscode-esp-idf-extension/blob/81685c07f047f3bf04aea4d049f411c7d4188652/src/extension.ts#L1304

**To Reproduce**
1. In VS Code, with the extension installed and set up for the current project, press F1 and select ESP-IDF: Open ESP-IDF Terminal.
2. Run `Get-Variable IDF_PATH* | Format-Wide` in the PowerShell terminal that appears.
3. You will see an empty variable named `IDF_PATH=C:\Users\user\esp-idf` instead of having a variable named `IDF_PATH` and a value of `C:\Users\user\esp-idf`

**Expected behavior**
It seems to work better if you select `cmd` as the default shell.
Perhaps the expected behavior should either be to properly detect and configure PowerShell, or perhaps the expected behavior should be to use `cmd.exe` rather than PowerShell in the first place.

**Environment (please complete the following information):**

- OS Name & Version: Windows 10 1909
- VSCode Version: 1.45.1
- ESP-IDF Version: 3.3.2
- Python Version: 3.8.2

pwmb, 2020-05-20:
@oakwhiz Thanks for such detailed bug report.

@brianignacio5 I think it would be better if we can support powershell as well, thoughts?? 

We must avoid to change the default terminal since for users this could cause frustration, if they prefer `proweshell` or something other than `cmd` (_if any_)

oakwhiz, 2020-05-20:
I have discovered a similar issue with the `PATH` variable - `Get-Variable Path* | Format-Wide` reveals an empty variable with an extremely long name.

brianignacio5, 2020-05-21:
I'm not sure the issue @oakwhiz  I tested with Powershell before `cmd.exe`

[image]

PR #107 (2020-05-21): add powershell env var terminal (VSC-332)
Fix Powershell command in ESP-IDF Terminal with proper env variable setting as shown in https://docs.microsoft.com/en-us/powershell/module/microsoft.powershell.core/about/about_environment_variables?view=powershell-7

Fix #106
closed 2020-05-21 by brianignacio5