espressif/vscode-esp-idf-extension #114: TypeError [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string. Received type undefined

syeedhaiderali, 2020-06-03:
Not able to install tools in vs code on a 32 bit machine.

***Screenshot
[image]

***Environment

OS Name: [Windows 7 32-bit]
VSCode Version: [e.g. 1.45.0]
ESP-IDF Version: [e.g. v4.0.1]
Python Version: [e.g. 3.8 32-bit]

brianignacio5, 2020-06-03:
**Logs**
If applicable, please share the log file which can be obtained from

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

**Additional context**
Add any other context about the problem here.

syeedhaiderali, 2020-06-03:
{"message":"Command failed: git describe --tags\nfatal: not a git repository (or any of the parent directories): .git\n","stack":"Error: Command failed: git describe --tags\nfatal: not a git repository (or any of the parent directories): .git\n\n\tat ChildProcess.exithandler (child_process.js:308:12)\n\tat ChildProcess.emit (events.js:203:13)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-06-03T05:56:54.318Z"}
{"message":"Command failed: reg query HKEY_LOCAL_MACHINE\\SOFTWARE\\WOW6432NODE\\PYTHON\nERROR: The system was unable to find the specified registry key or value.\r\n","stack":"Error: Command failed: reg query HKEY_LOCAL_MACHINE\\SOFTWARE\\WOW6432NODE\\PYTHON\nERROR: The system was unable to find the specified registry key or value.\r\n\n\tat ChildProcess.exithandler (child_process.js:308:12)\n\tat ChildProcess.emit (events.js:203:13)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-06-03T05:56:56.101Z"}
{"message":"Error looking for python in windows","stack":"Error: Command failed: reg query HKEY_LOCAL_MACHINE\\SOFTWARE\\WOW6432NODE\\PYTHON\nERROR: The system was unable to find the specified registry key or value.\r\n\n\tat ChildProcess.exithandler (child_process.js:308:12)\n\tat ChildProcess.emit (events.js:203:13)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-06-03T05:56:56.104Z"}
{"message":"WARNING: You are using pip version 19.2.3, however version 20.1.1 is available.\r\nYou should consider upgrading via the 'python -m pip install --upgrade pip' command.\r\n","stack":"Error: WARNING: You are using pip version 19.2.3, however version 20.1.1 is available.\r\nYou should consider upgrading via the 'python -m pip install --upgrade pip' command.\r\n\n\tat c:\\Users\\Haider\\.vscode\\extensions\\espressif.esp-idf-extension-0.2.2\\dist\\extension.js:1:6472\n\tat ChildProcess.exithandler (child_process.js:299:7)\n\tat ChildProcess.emit (events.js:203:13)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-06-03T07:08:58.129Z"}
{"message":"WARNING: You are using pip version 19.2.3, however version 20.1.1 is available.\r\nYou should consider upgrading via the 'python -m pip install --upgrade pip' command.\r\n","stack":"Error: WARNING: You are using pip version 19.2.3, however version 20.1.1 is available.\r\nYou should consider upgrading via the 'python -m pip install --upgrade pip' command.\r\n\n\tat c:\\Users\\Haider\\.vscode\\extensions\\espressif.esp-idf-extension-0.2.2\\dist\\extension.js:1:6472\n\tat ChildProcess.exithandler (child_process.js:299:7)\n\tat ChildProcess.emit (events.js:203:13)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-06-03T07:10:21.946Z"}
{"message":"WARNING: You are using pip version 19.2.3, however version 20.1.1 is available.\r\nYou should consider upgrading via the 'python -m pip install --upgrade pip' command.\r\n","stack":"Error: WARNING: You are using pip version 19.2.3, however version 20.1.1 is available.\r\nYou should consider upgrading via the 'python -m pip install --upgrade pip' command.\r\n\n\tat c:\\Users\\Haider\\.vscode\\extensions\\espressif.esp-idf-extension-0.2.2\\dist\\extension.js:1:6472\n\tat ChildProcess.exithandler (child_process.js:299:7)\n\tat ChildProcess.emit (events.js:203:13)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-06-03T08:57:08.403Z"}
{"message":"WARNING: You are using pip version 19.2.3, however version 20.1.1 is available.\r\nYou should consider upgrading via the 'python -m pip install --upgrade pip' command.\r\n","stack":"Error: WARNING: You are using pip version 19.2.3, however version 20.1.1 is available.\r\nYou should consider upgrading via the 'python -m pip install --upgrade pip' command.\r\n\n\tat c:\\Users\\Haider\\.vscode\\extensions\\espressif.esp-idf-extension-0.2.2\\dist\\extension.js:1:6472\n\tat ChildProcess.exithandler (child_process.js:299:7)\n\tat ChildProcess.emit (events.js:203:13)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-06-03T08:59:38.596Z"}
{"message":"WARNING: You are using pip version 19.2.3, however version 20.1.1 is available.\r\nYou should consider upgrading via the 'python -m pip install --upgrade pip' command.\r\n","stack":"Error: WARNING: You are using pip version 19.2.3, however version 20.1.1 is available.\r\nYou should consider upgrading via the 'python -m pip install --upgrade pip' command.\r\n\n\tat c:\\Users\\Haider\\.vscode\\extensions\\espressif.esp-idf-extension-0.2.2\\dist\\extension.js:1:6472\n\tat ChildProcess.exithandler (child_process.js:299:7)\n\tat ChildProcess.emit (events.js:203:13)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-06-03T09:07:37.138Z"}
[... cut ...] .1 is available.\r\nYou should consider upgrading via the 'python -m pip install --upgrade pip' command.\r\n\n\tat c:\\Users\\Haider\\.vscode\\extensions\\espressif.esp-idf-extension-0.2.2\\dist\\extension.js:1:6472\n\tat ChildProcess.exithandler (child_process.js:299:7)\n\tat ChildProcess.emit (events.js:203:13)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-06-03T09:08:21.690Z"}

brianignacio5, 2020-06-10:
Hi @syeedhaiderali Sorry for late reply. 

There is an issue with Ninja. Since [Ninja](https://github.com/ninja-build/ninja) is built using 64-bit binaries as shown in https://github.com/ninja-build/ninja/issues/1339 you will need to manually compile ninja in your machine. 

We did not capture this error and we should, sorry for this. 

After you compile Ninja, please follow [Onboarding doc](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/ONBOARDING.md) to manually configure the extension.

PR #121 (2020-06-15): Bugfix/idf tools not available (VSC-360)
throw error for IDF Tools when a version is not found. Capture errors on `downloadToolsInIdfToolsPath` caller.

Add cancellable gui menuconfig loading.

Should block IDF Tools download to fix #114 fo example.
closed 2020-06-15 by brianignacio5