espressif/vscode-esp-idf-extension #116: Add note to documentation about folderless workspaces?

bdpdx, 2020-06-04:
I had a situation yesterday where the extension would not run any commands.  No matter what was tried, VSCode would report "command not found."

After some digging, I found the error at line 65 in extension.ts:

`let workspaceRoot: vscode.Uri;`

in this case, `vscode.Uri` is `undefined` when the current workspace has no folders (this happens when a workspace is created).

There are bug reports in other projects of this being a problem so I think it's a VSCode issue rather than a vscode-esp-idf-extension problem, but it might be worth updating the onboarding doc to mention that the current workspace must have at least one folder or the extension will not work.

brianignacio5, 2020-06-05:
Can you show some steps to reproduce this behavior? 

Some commands can only work if some folder is open and there is usually validation for that case. But some command should work without an open folder, like `Configure ESP-IDF extension`

bdpdx, 2020-06-06:
My version of VSCode:

Version: 1.45.1
Commit: 5763d909d5f12fe19f215cbfdd29a91c0fa9208a
Date: 2020-05-14T08:33:47.663Z
Electron: 7.2.4
Chrome: 78.0.3904.130
Node.js: 12.8.1
V8: 7.8.279.23-electron.0
OS: Darwin x64 19.5.0

Installed extensions:

esp-idf-extension version 0.2.2, from Marketplace, enabled globally.
C/C++ 0.28.2
CMake Tools 1.4.0
Python 2020.5.80290

To see this bug:
 - Close all windows in VSCode
 - Create a new window File -> New Window
 - Save it somewhere as a workspace File -> Save Workspace As... "dir/foo.code-workspace"
 - Try to run a command
 - For example, ESP-IDF: Configure ESP-IDF extension

The command does not work and I see the following error:

[image]

As I mentioned in my original comment, I downloaded the source and ran it in the debugger.  The failure point is extension.ts line 65:

`let workspaceRoot: vscode.Uri;`

`vscode.Uri` is undefined, so workspaceRoot is undefined and the failures cascade from there.

PR #123 (2020-06-15): validate workspace folders length (VSC-362)
Fix #116 

PTAL @bdpdx
closed 2020-06-15 by brianignacio5