espressif/vscode-esp-idf-extension #118: Blink example with issues

versamodule, 2020-06-09:
Just installed this extension on a new PC today. Install went well. 
I selected the Blink example. It compiled fine, however when you open blink.c There are 2 issues in the "problems" tab. 

[image]

Also, I would like to see the compilation in progress, over just a popup. Is there a way to enable that ? 

[image]

brianignacio5, 2020-06-10:
The compilation process is shown in the Output tab-> ESP-IDF Build.

Did you configure the C/C++ extension as shown in [Configuration of c_cpp_properties.json file](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/C_CPP_CONFIGURATION.md)

versamodule, 2020-06-10:
Thanks for the Tip, that worked. 
Ya, i dont get anything in the output tab. I looked there first before mentioning it. 

[image]

brianignacio5, 2020-06-10:
There is a dropdown that says `Tasks`, change it to `ESP-IDF Build`

brianignacio5, 2020-06-11:
Closing this issue since it seems resolved.

closed 2020-06-11 by brianignacio5