espressif/vscode-esp-idf-extension #119: debugging wrover-kit-v4.1 with extension does not hit break point in tasks

Mair, 2020-06-10:
**Describe the bug**
when using the extensions to debug a ESP-WROVER-KIT-V4.1 I can start a debug session however if my code creates a task with breakpoints in it, the break points are not hit. If I use my own launch.json it does work

**To Reproduce**
create a project with the following code

```c
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

void tsk(void* params)
{
  while (true)
  {
    printf("Hello world!\n");
    vTaskDelay(pdMS_TO_TICKS(1000));
  }
}

void app_main(void)
{
  xTaskCreate(tsk,"tsk",1024 *5,NULL,5,NULL);
}
```
add a break point in the app_main as well as inside the tasks while loop

**Expected behavior**
both breakpoints should be hit

**Screenshots**
please see video link [here](https://vimeo.com/427655589/cbb76cce28)

**Environment **

- OS Name & Version: Windows 10
- VSCode Version: 1.45.1
- ESP-IDF Version: 4.0
- Python Version: 3.7

the launch.json that is wroking
```json
{
  // Use IntelliSense to learn about possible attributes.
  // Hover to view descriptions of existing attributes.
  // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
  "version": "0.2.0",
  "configurations": [
    {
      "name": "ESP32 OpenOCD",
      "type": "cppdbg",
      "request": "launch",
      "miDebuggerPath": "C:/esp/tools/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gdb.exe",
      "cwd": "${workspaceFolder}/build",
      //change the name of the elf file as required
      "program": "C:/projects/spike/wrover/debug/build/debug.elf",
      "setupCommands": [
        {
          "description": "Enable pretty-printing for gdb",
          "text": "-enable-pretty-printing",
          "ignoreFailures": true
        },
        {
          "text": "file 'C:/projects/spike/wrover/debug/build/debug.elf'"
        },
        {
          "text": "target remote 127.0.0.1:3333"
        },
        {
          "text": "set remote hardware-watchpoint-limit 2"
        },
        {
          "text": "monitor reset halt"
        },
        {
          "text": "flushregs"
        }
      ],
      "externalConsole": false,

      "logging": {
        // "trace": true,
        // "traceResponse": true,
        //"engineLogging": true
      }
    }
  ]
}

```

brianignacio5, 2020-06-10:
Can you provide more information ?
 There should be an output channel called `ESP-IDF Debug Adapter` and a `debug.log` is generated by the debug adapter in your current project (can see it in the video).

an-dr, 2020-06-16:
@brianignacio5 our debug adapter doesn't working because @Mair are using `"type": "cppdbg",` i.e. debugging goes through c/cpp extension, if I understand right. With `"type": "espidf",` that will be an issue related to the our debug adapter, not now I'm not sure how I can help here.

@Mair have you tried to debug with following `launch.json`? It is a bit less full of features now, but should work as expected

```
{
    "version": "0.2.0",
    "configurations": [
        {
            "type": "espidf",
            "name": "Launch",
            "request": "launch",
            "env": {
                    "PYTHONPATH": "${command:espIdf.getExtensionPath}/esp_debug_adapter/debug_adapter"
                }
        }
    ]
}

```

an-dr, 2020-06-24:
@Mair Okey, found it! The breakpoint is hitten here but the problem in that the thread wasn't switched to the thread with breakpoint
[image]

and at the same time 

[image]

@Mair can you confirm the same behaviour on your machine?

Mair, 2020-06-27:
I'm having trouble getting the extension working in my windows environment. 
I'll need to trouble shoot and get back to you

closed 2020-07-19 by github-actions[bot]