espressif/vscode-esp-idf-extension #125: Publish the extension on Open VSX

Leonardobat, 2020-06-14:
I'm a [VS Codium](https://github.com/VSCodium/vscodium) user, a FLOSS version of VS Code, it's the same code but without the Microsoft's telemetry and branding.
I want to request if possible to add the Espressif extension on the [open-vsx.org](https://open-vsx.org/) market to us doesn't need to works with the Microsoft marketplace as is unclear if it's legal to use it on VS Code forks, see [here](https://github.com/microsoft/vscode/issues/31168).

Additional info:
+ This is [how to](https://github.com/eclipse/openvsx/wiki/Publishing-Extensions) publish a new extension.
+ I know your extension relies on C++ Microsoft extension, one possible workaround would use the [Clangd ones](https://open-vsx.org/extension/llvm-vs-code-extensions/vscode-clangd).
+ I know too that your main compiler is GCC, but as there is an [LLVM port](https://github.com/espressif/llvm-project) in process and the clangd extension has no debug feature, would be only necessary to resolve all the `#include` stuff.

pwmb, 2020-06-14:
@Leonardobat did you tried to run the `.vsix` from the [Release Tab](https://github.com/espressif/vscode-esp-idf-extension/releases) and see if it works at all ?

Its true that we use c/c++ Extension from Microsoft as ESP-IDF's language server for intellisense but features like build/flash/monitor/menuconfig, etc., should work without the c/c++ extension itself.

If `.vsix` install fails, I would suggest you to build one `.vsix` after cloning the repo and modifying the `package.json`'s content

__From:__

```json
"extensionDependencies": [
    "ms-vscode.cpptools"
  ],
```

__To:__

```json
"extensionDependencies": [],
```

Steps to build vsix locally can be found [here](https://github.com/espressif/vscode-esp-idf-extension#build-from-source-code), but here is the quick steps for you:

- Clone this repo
- Install Nodejs
- Install `yarn`
- Run `yarn install` to install dependencies in the root of the cloned repo
- Run `yarn package` to generate a vsix file

Please do let us know your findings.

_Also a side note, if you want to remove the telemetry from vscode you can follow this [guide](https://code.visualstudio.com/docs/getstarted/telemetry)_

Leonardobat, 2020-06-14:
Thanks for the reply @pwmb.
Just install your `.vsix`, won't work as it is dependent on Microsoft's extension. Instead, build locally without the dependency, the building, flashing, monitor and the menu config features worked well.

pwmb, 2020-06-15:
Thanks @Leonardobat for testing this out... We will test release process to the `openvsx` marketplace, and let you know.

pwmb, 2020-09-09:
Since we still depends on `ms-vscode.cpptools ` extension for several tasks, can't release a fully functional extension in open-vsx marketplace.

In future once we have no dependency on the `ms-vscode.cpptools` extension we could proceed with the release, till then will close this ticket.

closed 2020-09-09 by pwmb