espressif/vscode-esp-idf-extension #131: Strange Output

mitchjs, 2020-06-19:
windows 10 x64 1909
vs code 1.46.1
idf-ext 0.3.0

so, just got the new version, as it updated itself
and did a build of my existing project
and noticed there is no more build output.. it is now in terminal
as its launching a task.. which is fine
but i did notice this in the build output - tasks

Error: the task 'espressif.esp-idf-extension: ESP-IDF Compile' neither specifies a command nor a dependsOn property. The task will be ignored. Its definition is:
{
    "type": "shell",
    "id": "shell,cmake -G Ninja ..,",
    "problemMatcher": [
        "idfRelative",
        "idfAbsolute"
    ],
    "label": "espressif.esp-idf-extension: ESP-IDF Compile"
}

everything builds just fine.. just noticed that message...mean anything ??

thanks for the hard work on this extension!

brianignacio5, 2020-06-19:
Hey @mitchjs Thanks for the feedback. We moved build and flash as tasks to capture errors on the problems tab instead of the output as before.

Will take a look on this message.

mitchjs, 2020-06-19:
also, i just noticed this message in the build output(terminal)

```
CMake Warning at C:/Users/mitchjs/esp-idf-v4.0.1/tools/cmake/idf.cmake:14 (message):
  IDF_PATH environment variable is different from inferred IDF_PATH.

                              Check if your project's top-level CMakeLists.txt includes the right
                              CMake files. Environment IDF_PATH will be used for the build.
Call Stack (most recent call first):
  C:/Users/mitchjs/esp-idf-v4.0.1/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:5 (include)
```

brianignacio5, 2020-06-23:
The second message is due to the extension modifying its own environment variables. A solution would be to use system-wide IDF_PATH or replace it, but that would limit the ability for the extension to configure different esp-idf versions or require to modify your system.

mitchjs, 2020-06-23:
brian, i dont fully understand...

if in vscode i do ESP-IDF: Configure paths, ESP_IDF is c:\Users\mitchjs\esp-idf-v4.0.1
which is what i would expect it to be
and as far as i know, i have no system wide IDF_PATH enviroment var
i go to cmd prompt type SET, and dont see one

its more of a WARNING? than ERROR?
everything seems to build just fine

PR #135 (2020-06-28): add idf task definition (VSC-374)
Not really an error but was using predefined shell task definition which required some additional fields and had some output in output tasks.

Instead, defined our esp-idf task definition.

Fix #131
closed 2020-06-28 by brianignacio5