espressif/vscode-esp-idf-extension #132: Debugging broke in 0.3.0

meltdown03, 2020-06-21:
After upgrading to 0.3.0, debugging doesn't start. I don't get any errors, I just see the debug controls at the top for a fraction of a second and then disappear. OpenOCD and Debug Adapter don't start. Even setting debug type to cppdbg or MI in launch.json don't work unless I revert to the last version of vscode-esp-idf-extension. I re-ran the configuration, still nothing.
Here is log file:
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/4808745/esp_idf_vsc_ext.log)

Monkeyinapocket, 2020-06-22:
I can confirm this issue. It appears on Windows and Linux machines.

venon1983, 2020-06-23:
Yes, I have same trouble. Because this problem, I had to make downgrade.

brianignacio5, 2020-06-23:
On launch.json, can you try changing:

`"PYTHONPATH": "${command:espIdf.getExtensionPath}/esp_debug_adapter/debug_adapter"`

to

`"PYTHONPATH": "**ESP_IDF_EXTENSION_FOLDER**/esp_debug_adapter/debug_adapter"`

Or please try #134 and see if it fixes the issue. For some reason it is not resolving the command.

venon1983, 2020-06-23:
Thaks, it was helpful. Now looks everything correct.

PR #134 (2020-06-23): fix args for debug adapter update doc (VSC-373)
Fix #132 

For some reason ${command:espIdf.getExtensionPath} is failing in launch.json. Removed from there and added within debugAdapterManager configureWithDefaultValues method.

Fixed other small things like fire(null) and default values for adapter args.
closed 2020-06-23 by brianignacio5