espressif/vscode-esp-idf-extension #133: Menuconfig fails to actually save changes

mitchjs, 2020-06-22:
ESP-IDF = MASTER
Extension = 0.3.0
VS Code = 1.46.1
OS = Win 10 x64 1909

after having menu config open and doing a few builds
i go to make a change in menu config, and click save
and it doesnt actually save, never shows the lower left side
notification popup that it saved

in the output window just shows:
**{"version":2,"save":"c:\\esp32\\bleprph\\sdkconfig"}**
when i click on save
to get it to work, i have to CLOSE VSCODE, reopen and relaunch menuconfig gui

then it shows in output window when i click save
**{"version":2,"save":"c:\\esp32\\bleprph\\sdkconfig"}
{"version": 2, "values": {}, "ranges": {}, "visible": {}}**

thanks all for great work!

brianignacio5, 2020-06-23:
Hi @mitchjs Thanks for the feedback ! I tried gui menuconfig and it is saving the values.

Please give us more details about your setup and steps to reproduce. (For example, which values did you try to modify)

mitchjs, 2020-06-23:
Yes it does save most of the time... it seems after sometime doing builds it stops saving... doesn’t matter which setting ... it’s like the file is locked maybe
I can’t reproduce at will... it just happens

brianignacio5, 2020-06-23:
Can try to see if there is something in the log

**Logs**
If applicable, please share the log file which can be obtained from

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

mitchjs, 2020-06-23:
will do next time it fails...

mitchjs, 2020-07-13:
i havent had it happen, so closing, if it does happen i dig deeper

closed 2020-07-13 by mitchjs