espressif/vscode-esp-idf-extension #136: Launch GUI configuration tool and monitor from taskbar not working in VSCode

Aietes, 2020-06-30:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [ ] Make sure you have searched for existing bugs and features request before you post an issue.
- [ ] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [ ] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
A clear and concise description of what the bug is (current behavior).

Launching the GUI configuration tool and monitor from taskbar is not working, and reports a Python problem. This is misleading though, as both menuconfig and monitor work from the terminal, meaning all Python requirements are met. This seems to be a problem with the VSCode Extension.

The following Python requirements are not satisfied:

gdbgui>=0.13.2.0

pygdbmi<=0.9.0.2

reedsolo==1.5.3

bitstring>=3.1.6

esp-windows-curses; sys_platform == 'win32'

To install the missing packages, please run "C:\Users\Stefan\Development\esp-idf\install.bat"

Diagnostic information:

    IDF_PYTHON_ENV_PATH: .

    Python interpreter used: C:\Users\Stefan\.platformio\penv\Scripts\python.exe

ESP-IDF v4.2-dev-1905-g625bd5eb1

---------------------------ERROR--------------------------

Received signal: null with code: 1
-----------------------END OF ERROR-----------------------

**To Reproduce**
Steps to reproduce the behavior (including the full log).

Click on "ESP-IDF Monitor device" or "ESP-IDF Launch GUI configuration tool" in the taskbar, or launch from main menu.

**Expected behavior**
A clear and concise description of what you expected to happen.

I'd expect menuconfig and monitor to launch.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

- OS Name & Version: [e.g. MacOS X 10.14.5] Windows 10 release 2004
- VSCode Version: [e.g. 1.35.1] latest stable
- ESP-IDF Version: [e.g. v3.2.2] latest stable
- Python Version: [e.g. 2.7.10] 2.8.3

**Logs**
If applicable, please share the log file which can be obtained from

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

**Additional context**
Add any other context about the problem here.

brianignacio5, 2020-06-30:
Did you configure the extension as shown in [Onboarding documentation](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/ONBOARDING.md) ?

The extension has some configuration settings used to determine IDF_PATH, the python virtual environment to use, Tools locations, etc as explained in [README](https://github.com/espressif/vscode-esp-idf-extension#esp-idf-configure-extension). Please make sure these values are correct.

Aietes, 2020-06-30:
Yes, these are all set and should be correct. Since the build tasks are working fine, and are using python as well. There is no setting for the python environment in the VS Code ESP-IDF extension settings, maybe that's where the error comes from? It seems: IDF_PYTHON_ENV_PATH: .   may be the culprit, but I'm not aware where to configure this variable.

brianignacio5, 2020-07-01:
Does your project has a `.vscode` folder with a `settings.json` that could be overriding user settings or workspace settings ?

Aietes, 2020-07-01:
I do have a `.vscode` folder with a `settings.json`, but there is nothing in there regarding `idf.pythonBinPath`. It's set to the default `python` in the user settings.

The extension configuration as posted by you seems to be working for me to some extent, as it seems clicking on the `ESP-IDF Monitor device` produces the right commands:

```
PS C:\Users\Stefan\Development\void-safe-esp-idf> set IDF_PATH=C:\Users\MyUser\Development\esp-idf
PS C:\Users\Stefan\Development\void-safe-esp-idf> python C:\Users\MyUser\Development\esp-idf\tools\idf.py -p COM3 monitor
```

But it still produces this error directly afterwards:

```
The following Python requirements are not satisfied:
gdbgui>=0.13.2.0
pygdbmi<=0.9.0.2
reedsolo==1.5.3
bitstring>=3.1.6
esp-windows-curses; sys_platform == 'win32'
To install the missing packages, please run "C:\Users\MyUser\Development\esp-idf\install.bat"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: .
    Python interpreter used: C:\Users\MyUser\.platformio\penv\Scripts\python.exe
```

The last two lines don't look right to me, as they are not reflecting the python environment or bin path.

mitchjs, 2020-07-02:
@Aietes, do you have the platform.io extension disabled?
i used to use it for ESP32 dev, and just disabled it and now using this extension exclusively
maybe try disable, and then relaunch the onboarding process

brianignacio5, 2020-07-02:
The thing, when you install ESP-IDF, there is virtual environment python created in some location like C:\Users\myUser\.espressif\python_env\idf4.2_py3.7_env\Scripts\python.exe which is used as Python Interpreter for ESP-IDF (when you use ESP-IDF Terminal for example). In this virtual environment, ESP-IDF requirements are installed.

From what you are posting there are two issues with your setup:

1) You have the default python value. This means our extension scripts will use whatever python executable they can find in the environment PATH variable, which seems to be picking the one from PlatformIO extension, since they add it to the vscode environment variable PATH. Since idf.pythonBinPath is set to `python`, the IDF_PYTHON_ENV_PATH fails since it has no parent directory. These packages are not installed in C:\Users\MyUser\.platformio\penv\Scripts\python.exe

2) You haven't configured the extension. Press F1 -> type Configure ESP-IDF extension. Follow the steps in [Onboarding Documentation](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/ONBOARDING.md). This will save all the required values in the extension configuration settings such as idf.espIdfPath and idf.pythonBinPath

Aietes, 2020-07-02:
Thank you for your help! I changed the `Idf: Python Bin Path Win` in the VS Code preferences to:
```
C:\Users\MyUser\.espressif\python_env\idf4.2_py3.8_env\Scripts\python.exe
```
Problem solved! Thank you for the great work on the extension!

closed 2020-07-02 by Aietes

unkpers, 2020-11-29:
I have checked all of those and I couldn't find the problem :

idf_versions.txt has (1) KB
idf_versions.txt progress: 100.00% (0.08 / 0.08) KB
Saving C:\Users\u\AppData\Local\Programs\Python\Python37\python.exe to create python virtual environment.

Downloading Toolchain for Xtensa (ESP32) based on GCC
Found xtensa-esp32-elf in c:\Users\u\esp\esp-idf\tools\dist
Downloading Toolchain for Xtensa (ESP32-S2) based on GCC
Found xtensa-esp32s2-elf in c:\Users\u\esp\esp-idf\tools\dist
Downloading Toolchain for ESP32 ULP coprocessor
Found esp32ulp-elf in c:\Users\u\esp\esp-idf\tools\dist
Downloading CMake build system
Error: Cannot delete files/folders outside the current working directory. Can be overridden with the `force` option.
The following Python requirements are not satisfied:
click>=5.0
pyserial>=3.0
future>=0.15.2
cryptography>=2.1.4
pyparsing>=2.0.3,<2.4.0
pyelftools>=0.22
gdbgui==0.13.2.0
pygdbmi<=0.9.0.2
kconfiglib==13.7.1
reedsolo>=1.5.3,<=1.5.4
bitstring>=3.1.6
ecdsa>=0.16.0
esp-windows-curses; sys_platform == 'win32'
To install the missing packages, please run "C:\Users\u\esp\esp-idf\install.bat"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: .
    Python interpreter used: C:\Users\u\AppData\Local\Programs\Python\Python37\python.exe
Command failed: "C:\Users\u\AppData\Local\Programs\Python\Python37\python.exe" "C:\Users\u\esp\esp-idf\tools\check_python_dependencies.py" -r "C:\Users\u\esp\esp-idf\requirements.txt"

brianignacio5, 2020-11-30:
@unkpers What is the value of `Idf: Python Bin Path Win` in your Visual Studio Code settings ?

unkpers, 2020-11-30:
"idf.pythonBinPath": "PYTHON_INTERPRETER"