espressif/vscode-esp-idf-extension #150: Subfolder

COM8, 2020-07-29:
**Is your feature request related to a problem? Please describe.**
My project structure looks like the following:
```
project/
  +--> esp32/
    +--> CMakeList.txt
    +--> main/
      +--> ...
    +--> ...
  +--> docs/
  +--> ..
```
I'm opening the `project` folder with vscode and all the ESP32 related stuff is inside the `esp32` (`CMakeList.txt`, ...).
Is it possible to configure a subfolder as base path for the `idf.py` actions. 

**Describe the solution you'd like**
Allow setting a subfolder as project root.

**Describe alternatives you've considered**
Open only the subfolder `esp32`.

brianignacio5, 2020-07-30:
Have you tried using Visual Studio Code Workspace projects ?

On [Working with multiple projects section in the README](https://github.com/espressif/vscode-esp-idf-extension#working-with-multiple-projects) we explain how we leverage the `.code-workspace` file to work with multiple projects.  As explain in [Visual Studio Code multi-root workspaces](https://code.visualstudio.com/docs/editor/multi-root-workspaces#_workspace-file-schema) you can create a new workspace and add project folder in this workspace. Using our command `Pick a workspace folder`.

COM8, 2020-07-30:
@brianignacio5 No, I did not knew about this.
This solves my problem. Thanks!

closed 2020-07-30 by COM8