espressif/vscode-esp-idf-extension #152: should/can we get other versions of ESP32-IDF

mitchjs, 2020-08-08:
the onboarding process... 

i know i can MANUALLY clone IDF into folders, but should the onboarding process be able to do more options other than

4.0.1
3.3.2
release/4.0.0
release/3.3.3
master

is there a way to update the list from: https://github.com/espressif/esp-idf/releases
and maybe all "release/xxxx"

also maybe some way to know/see what IDF version you are using right in the IDE
i have multiple branches on my machine
*im not even sure how to tell just by looking at the contents of the ESP-IDF folder
i thought idf.py --version, but that just seems to return 1.0.1

thanks
mitch

brianignacio5, 2020-08-10:
Hi @mitchjs the thing is, we have defined a [ESP-IDF Support Policy](https://github.com/espressif/esp-idf/blob/master/SUPPORT_POLICY.md) where we explained that we have some Long Term Support (LTS) versions and some unreleased version (master) which we currently support and older versions we no longer support actively. 

You can of course use any ESP-IDF version you like, but for example, the extension have many features that only work with ESP-IDF v4.0 or greater such as GUI Menuconfig (confserver.py was introduced after v3.3), CMake build system, etc.

You can also try `git describe --tags` in your ESP-IDF to check the version of your current folder.

mitchjs, 2020-08-12:
@brianignacio5 , yes i would expect only versions that this supports, ie v4.1 which is at RC now

brianignacio5, 2020-08-13:
> Pre-release versions (betas, previews, -rc and -dev versions, etc) are not covered by any support period.

brianignacio5, 2020-08-27:
Closing this issue since no additional work is expected here.

closed 2020-08-27 by brianignacio5