espressif/vscode-esp-idf-extension #153: Flash&Monitor tasks/actions from bottom bar do not work

michalbielecki, 2020-08-10:
**Describe the bug**
After successfully configuring the addon and forcing it to work with esp-idf release v4.1 I've noticed that only "build project action is working as expected".  "Flash device" and "Monitor device" does not work. Looks like there is some slash/backslash platform dependent problem when launching from bottom bar. Launching tasks from "Terminal -> Run Tasks" ends with pop-up message: "There is no task provider registered for tasks of type "esp-idf".

Launching from bottom bar generated following log:

- Flash:
`> Executing task in folder sources: c:\_Tools\ESP\esp32\python_env\idf4.1_py3.8_env\Scripts\python.exe c:\_Tools\ESP\esp-idf\components\esptool_py\esptool\esptool.py -p COM5 -b 115200 --after hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size detect 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 GardenController.bin <

/usr/bin/bash: c:_ToolsESPesp32python_envidf4.1_py3.8_envScriptspython.exe: command not found
The terminal process "C:\_Tools\Git\bin\bash.exe '-c', 'c:\_Tools\ESP\esp32\python_env\idf4.1_py3.8_env\Scripts\python.exe c:\_Tools\ESP\esp-idf\components\esptool_py\esptool\esptool.py -p COM5 -b 115200 --after hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size detect 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 GardenController.bin'" failed to launch (exit code: 127).

Terminal will be reused by tasks, press any key to close it.`

- Monitor:
`SP\esp-idf\tools\idf.py -p COM5 monitor_py3.8_env\Scripts\python.exe c:\_Tools\ES
bash: c:_ToolsESPesp32python_envidf4.1_py3.8_envScriptspython.exe: command not found`

- OS Name & Version: **Windows 10 19041.388**
- VSCode Version: **1.47.3**
- ESP-IDF Version: **release/v4.1** - also doesn't work with latest release suggested by the addon
- Python Version: **3.8.5**

michalbielecki, 2020-08-10:
I guess it can be rejected as not an issue. I've noticed that the outcome is depending on default shell used in VSCode. On daily basis, I'm using Git bash. Setting "terminal.integrated.automationShell.windows": "C:\\Windows\\System32\\cmd.exe" forces tasks and addons to use regular Windows CMD.

brianignacio5, 2020-08-11:
What is the value of your `idf.pythonBinPath` ? Go to Visual Studio Code Settings UI (F1 and type settings ui) and find `Python absolute binary path used to execute ESP-IDF Python Scripts` and see the the value there.

**Logs**
If applicable, please share the log file which can be obtained from

Windows: %USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log
Linux & MacOSX: $HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log

closed 2020-09-01 by github-actions[bot]

wenhuafeng, 2022-04-09:
/usr/bin/bash: D:Espressifpython_envidf4.4_py3.8_envScriptspython.exe: command not found
终端进程“D:\program files\Git\bin\bash.exe '-c', 'D:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe D:\Users\wenhuafeng\esp\esp-idf\tools\idf_size.py d:\Users\wenhuafeng\esp\esp-idf\examples\get-started\hello_world\build\hello_world.map'”已终止，退出代码: 127。