espressif/vscode-esp-idf-extension #154: IDF Onboarding stuck on verify idf Tools

jakubpav, 2020-08-10:
Hello, 
I've got a little problem with IDF Onboarding. 
It's not first time I'm setting this but this error has never crossed my way.

First I've manually installed esp-idf and idf-tools regarding [https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html](url)
Then I've downloaded Espressif IDF from marketplace.
Everything looked fine until the verification of idf tools where it writes `python is not accesible.` But I'm sure that python packages are all right.  

[image]

I'm not sure whats wrong so if somebody could guide me? 

thanks
Jakub Pavelka

brianignacio5, 2020-08-11:
Which Python did you choose on the onboarding window?

Please provide more information of your setup such as:

**Environment (please complete the following information):**

- OS Name & Version: [e.g. MacOS X 10.14.5]
- VSCode Version: [e.g. 1.35.1]
- ESP-IDF Version: [e.g. v3.2.2]
- Python Version: [e.g. 2.7.10]

**Logs**
If applicable, please share the log file which can be obtained from

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

jakubpav, 2020-08-11:
- OS Name & Version: openSUSE Tumblweed (Kernel 5.7.1)
- VSCode Version: 1.47.3
- ESP-IDF Version: v4.3
- Python Version: 3.8.4

[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/5054910/esp_idf_vsc_ext.log)

brianignacio5, 2020-08-12:
Please follow the steps from https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/ONBOARDING.md#installed-esp-idf-following-the-espressif-documentation-and-want-to-reuse-it-in-the-extension when you have preinstalled ESP-IDF and IDF Tools using `IDF_PATH/install.sh`

jakubpav, 2020-08-12:
I followed. It wasn't still working. It was the python path.
I've python on `/usr/bin/python`
But the environment and python for IDF was at `~/.espressif/python_env/idf4.3_py3.8_env/bin/python`
Thanks for helping me out! :)

closed 2020-08-12 by jakubpav