espressif/vscode-esp-idf-extension #158: [Bug Report]: Hard coded path in setup causes failure when trying to debug esp32

Onuonga, 2020-08-31:
**Describe the bug**
Upon trying to setup the esp32 with a debugger a user will run into this error attached in the screenshot. 
It seems there is some hard linking being done and it is then distributed to multiple user machines. 

**To Reproduce**
Setup according to the instructions on the repository. 

In my case, my launch.json is: 
```
{
	"version": "0.2.0",
	"configurations": [	
		{
			"name": "Debug",
			"type": "cppdbg",
			"request": "launch",
			"program": "${workspaceFolder}/build/esp-http-client-example.elf",
			"cwd": "${workspaceFolder}",
			"miDebuggerPath": "xtensa-esp32-elf-gdb",
			"preLaunchTask": "OpenOCD",
            "setupCommands":[{
                "text": "target remote :3333",
            },{
                "text": "mon reset halt",
            },{
                "text": "flushregs",
            },{
				"text": "thb http_call_task"
			}],
            "externalConsole": true
		}]
}

```

My settings.json is: 
```
{
    "cmake.configureOnOpen": false,
    "idf.espIdfPath": "<path to the repo>/esp/esp-idf",
    "idf.toolsPath": "<path to the repo>/.espressif",
    "idf.pythonBinPath": "<path to the repo>/.espressif/python_env/idf4.3_py3.8_env/bin/python",
    "idf.customExtraPaths": "<path to the repo>/.espressif/python_env/idf4.3_py3.8_env/bin:<path to the repo>/.espressif/python_env/idf4.0_py3.8_env/bin:/<path to the repo>/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin:<path to the repo>/.espressif/tools/xtensa-esp32s2-elf/esp-2020r2-8.2.0/xtensa-esp32s2-elf/bin:/<path to the repo>/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/<path to the repo>/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/<path to the repo>a/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin",
    "idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"/<path to the repo>/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/share/openocd/scripts\"}",
    "idf.openOcdConfigs": [
        "board/esp32-wrover-kit-3.3v.cfg "
    ],
    "cmake.defaultVariants": {
        "buildType": {
            "default": "debug",
            "description": "The build type.",
            "choices": {
                "debug": {
                    "short": "Debug",
                    "long": "Disable optimizations - include debug information.",
                    "buildType": "Debug"
                },
                "release": {
                    "short": "Release",
                    "long": "Optimize for speed - exclude debug information.",
                    "buildType": "Release"
                },
                "minsize": {
                    "short": "MinSizeRel",
                    "long": "Optimize for smallest binary size - exclude debug information.",
                    "buildType": "MinSizeRel"
                },
                "reldeb": {
                    "short": "RelWithDebInfo",
                    "long": "Optimize for speed - include debug information.",
                    "buildType": "RelWithDebInfo"
                }
            }
        }
    },
    "python.pythonPath": "/<path to the repo>/.espressif/python_env/idf4.3_py3.8_env/bin/python",
    "C_Cpp.default.includePath": [
        "**<path to the repo>**/esp/esp-idf/examples/common_components/protocol_examples_common/",
        "${default}"
    ],
    "C_Cpp.dimInactiveRegions": false,
}
```
My tasks.json is: 
```
{
    "version": "2.0.0",
    "tasks": [
      {
        "label": "OpenOCD",
        "type": "shell",
        "isBackground": true,
        "command": "openocd -f ~/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/share/openocd/scripts/board/esp32-wrover-kit-3.3v.cfg",
        "problemMatcher": [
          {
            "pattern": [
              {
                "regexp": ".",
                "file": 1,
                "location": 2,
                "message": 3
              }
            ],
            "background": {
              "activeOnStart": true,
              "beginsPattern": ".",
              "endsPattern": ".",
            }
          }
        ]
      }
    ]
  }
```

**Expected behavior**
it should not be doing this and instead this should be a relative path based on the user who installed the extension. 

**Screenshots**

[image]

**Environment (please complete the following information):**

- OS Name & Version: [e.g. MacOS X 10.15.6]
- VSCode Version: [e.g. 1.48.2]
- ESP-IDF Version: [e.g. v4.3]
- Python Version: [e.g. 3.8]

Note: All ^<path to the repo>^above will need to be replaced on your machine.

brianignacio5, 2020-09-03:
You are using the [Microsoft C/C++ Extension Debug Adapter](https://github.com/microsoft/vscode-cpptools) which is not supported by us. Can you try using our debug adapter or post an issue in their GitHub repository ?

Onuonga, 2020-09-03:
Thanks for the tip @brianignacio5. 
I am not too clear on what you mean by your debug adapter, kindly clarify. 
I am going to try assume you mean ```xtensa-esp32-elf-gdb```. Is this correct ? 
 
Just to add, I did try follow the same steps in 3 terminal sessions where in terminal: 
1. I have the monitor running via ``ìdf.py monitor```
2. I am running openocd 
3. I run ```xtensa-esp32-elf-gdb``` (file set to the elf file generated during build) but I seem to run into the same error while debugging. 
[image]

brianignacio5, 2020-09-04:
HI @Onuonga The thing is, vscode uses the [Debug Adapter Architecture](https://code.visualstudio.com/api/extension-guides/debugger-extension#debugging-architecture-of-vs-code) to allow any extension to supply their own debug server (a middle man between GDB and vscode to parse messages between them). Your **launch.json** uses _cppdbg_ which is the Debug Adapter implemented by the [Microsoft C/C++ Extension](https://github.com/microsoft/vscode-cpptools) which we do not maintain. 

If you try using our Debug Adapter (by following configuration in [Debugging](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/DEBUGGING.md)) we can actually make changes for this debug scenario.

Onuonga, 2020-09-08:
Sure thing. 
Thank you for the info @brianignacio5 
I will make change the adapter and share my progress.

brianignacio5, 2020-09-22:
Will close this issue since it is not related to our extension.

Feel free to open a new issue if you face any problems.

closed 2020-09-22 by brianignacio5